pycom/pycom-micropython-sigfox #1: int32 overflow because of to late cast

livius2, 2017-03-28:
int32 overflow because of to late cast
```
STATIC mp_obj_t machine_deepsleep (uint n_args, const mp_obj_t *arg) {
    mperror_enable_heartbeat(false);
    bt_deinit(NULL);
    wlan_deinit(NULL);
    if (n_args == 0) {
        esp_deep_sleep_start();
    } else {
        esp_deep_sleep((uint64_t)(mp_obj_get_int(arg[0]) * 1000));
    }
}
```

but should be

```
 esp_deep_sleep((uint64_t)mp_obj_get_int(arg[0]) * 1000);
```

jmarcelino, 2017-03-28:
Discussed in forum thread https://forum.pycom.io/topic/944/machine-sleep-time_in_millis-does-time_in_millis-has-some-limit-not-works-for-6-hours-value

danicampora, 2017-06-02:
Already solved, thanks @livius2 and @jmarcelino

closed 2017-06-02 by danicampora