pycom/pycom-micropython-sigfox #4: Old Issues

robert-hh, 2017-04-03:
Please consider also the open issues from the pycom-micropython branch.
They should not get lost. Most of them are also applicable to this branch.

danicampora, 2017-04-04:
@robert-hh thanks, we haven't forgotten about the issues on the other repo. They will be solved shortly.

danicampora, 2017-05-24:
@me-no-dev: When we met the other week you mentioned me a trick to remove all the print logs from the IDF. Can you tell me how? Thanks :-)

me-no-dev, 2017-05-24:
Hi Daniel :) Nice to "see" you again! Here is the line that you need to add as first thing in app_main : https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/esp32-hal-misc.c#L82
This should help with hiding most of the messages

robert-hh, 2017-05-24:
This line suppressed most of the messages:
`esp_log_level_set("*", ESP_LOG_NONE);`
and I had to add also
`#include "esp_log.h"`
That's what remained:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff9010,len:8
load:0x3fff9018,len:248
load:0x40078000,len:4056
load:0x4009fc00,len:920
entry 0x4009fde4
```

me-no-dev, 2017-05-24:
The messages that remain come from the bootloader in ROM and can not be suppressed by software.
I think GPIO12 was the one that is used to suppress those messages.
The messages that the line suppressed were coming from the libs that are in binary form, so they do not follow the log level set in IDF.

closed 2018-01-25 by robert-hh