pycom/pycom-micropython-sigfox #6: LoPy build error on linux 64 machine

saurubh, 2017-05-02:
Building LoPy(Latest firmware update today) from Pycom. Getting this error on linux 64 bit OS. Please resolve this. Even i have tried the method above but no luck.

Error shown below-----

skmis@skmis-VirtualBox:~/esp/esp32$ make BOARD=LOPY -j5 LORA_BAND=USE_BAND_868 TARGET=app
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LINK build/LOPY_868/release/application.elf
/home/skmis/esp/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lrtc_clk
collect2: error: ld returned 1 exit status
application.mk:360: recipe for target 'build/LOPY_868/release/application.elf' failed
make: *** [build/LOPY_868/release/application.elf] Error 1

Xykon, 2017-05-11:
I'm not getting this error. I suggest you try make BOARD=LOPY -j5 LORA_BAND=USE_BAND_868 TARGET=app V=1 and post the complete output.
Maybe your pycom-esp-idf needs an update?

sdaniel55, 2018-09-03:
Hi 👋 
I'll close this issue because it has not had any recent activity. Thank you for your contributions!

closed 2018-09-03 by sdaniel55