pycom/pycom-micropython-sigfox #8: Heartbeat doesn't work while code is running

DvdGiessen, 2017-05-15:
Platform: `(sysname='LoPy', nodename='LoPy', release='1.6.13.b1', version='v1.8.6-607-g9c8a0e9e on 2017-05-01', machine='LoPy with ESP32', lorawan='1.0.0')`

**Expected behaviour, quoted from documentation:**
 > By default the heartbeat LED flashes in blue color once every 4s to signal that the system is alive.

**Observed behaviour:**
The heartbeat only flashes when there's no code running!

**Steps to reproduce:**
1. `os.mkfs('/flash')`, to make sure we're in a completely clean state.
2. Reset the board.
4. Observe that heartbeat LED flashes every 4 seconds.
3. Run from the REPL:
   ```py
   while True:
       pass
    ```
4. Observe that heartbeat LED doesn't flash anymore.

**Alternative ways to reproduce:**
Different `main.py` files using busy loops, machine idle loops, time sleep loops:
```py
if __name__ == '__main__':
    while True:
        pass
```
```py
import machine
if __name__ == '__main__':
    while True:
        machine.idle()
```
```py
import utime
if __name__ == '__main__':
    while True:
        utime.sleep_ms(10)
```

`main.py` file having running code in a separate thread:
```py
import _thread
def foo():
    while True:
        pass
if __name__ == '__main__':
    _thread.start_new_thread(foo, ())
```

See also [the forum thread](https://forum.pycom.io/topic/1216/heartbeat-doesn-t-work-while-code-is-running) at the Pycom forums.

oligauc, 2017-07-06:
Your infinite loop is consuming 100% of CPU resources. Try

import time
if __name__ == '__main__':
    while True:
        time.sleep(5)

nevercast, 2017-12-05:
@oligauc's answer is the right.

The heartbeat LED is lower priority than your main task. The main task is never blocked (asleep) so that another task (heartbeat) can do work.

You should only run code as often as needed, I don't believe micropython offers thread priorities yet so you will need to sleep. the utime module offers sleep, sleep_ms and sleep_us. 

In future it would be nice to be able to change the priority (and name) of threads (tasks).

nevercast, 2018-01-22:
Now that we have dual core support, the heart beat could maybe move to the WiFi/BLE core.

danicampora, 2018-01-22:
@nevercast good idea. I'll take a look at it. Thanks!