pycom/pycom-micropython-sigfox #15: BLE - OSError: connection refused

Bucknalla, 2017-05-31:
```
(sysname='LoPy', nodename='LoPy', release='1.6.13.b1', version='v1.8.6-607-g9c8a0e9e on 2017-05-01', machine='LoPy with 
ESP32', lorawan='1.0.0') 
```
Attempting to connect from one LoPy to another LoPy device:

```Python
from network import Bluetooth
import binascii
bluetooth = Bluetooth()

# scan until we can connect to any BLE device around
bluetooth.start_scan(-1)
adv = None
while True:
    adv = bluetooth.get_adv()
    if adv:
        try:
            print(binascii.hexlify(adv.mac))
            bluetooth.connect(adv.mac)
        except:
            # start scanning again
            bluetooth.start_scan(-1)
            continue
        break
print("Connected to device with addr = {}".format(binascii.hexlify(adv.mac)))
```

Error arises after the connecting LoPy finds the address of the advertising device and attempts to connect.

```
b'240ac40061fa'
Traceback (most recent call last):                                                                                      
  File "<stdin>", line 50, in <module>                                                                                  
OSError: connection already closed
```
                                                                         
Thanks!

grandcat, 2017-06-01:
Maybe try `bluetooth.init()` after `bluetooth = Bluetooth()`

danicampora, 2017-06-02:
Solved on the latest release.

closed 2017-06-02 by danicampora