pycom/pycom-micropython-sigfox #17: Build failure on OSX

jeremyherbert, 2017-06-07:
When building for LoPy: `make BOARD=LOPY LORA_BAND=USE_BAND_915 TARGET=app`

I needed to do this first: https://github.com/micropython/micropython/commit/326343feebdd1ad990864cc830a30b6a44d44918

jeremyherbert, 2017-06-07:
I'm now stuck at this error:

```
CC mods/machtimer_alarm.c
In file included from /Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:20:0,
                 from /Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/driver/include/driver/timer.h:19,
                 from mods/machtimer_alarm.c:6:
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLUP_DIS':
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/esp32/include/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:169:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_CLR_BIT);                                                                    \
             ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:96:5: note: in expansion of macro 'REG_CLR_BIT'
     REG_CLR_BIT(PIN_NAME, FUN_PU);
     ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/esp32/include/esp_assert.h:23:28: error: expression in static assertion is not constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:169:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_CLR_BIT);                                                                    \
             ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:96:5: note: in expansion of macro 'REG_CLR_BIT'
     REG_CLR_BIT(PIN_NAME, FUN_PU);
     ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLUP_EN':
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/esp32/include/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:101:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PU);
     ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/esp32/include/esp_assert.h:23:28: error: expression in static assertion is not constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
                            ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:138:38: note: in expansion of macro 'TRY_STATIC_ASSERT'
 #define ASSERT_IF_DPORT_REG(_r, OP)  TRY_STATIC_ASSERT(!IS_DPORT_REG(_r), (Cannot use OP for DPORT registers use DPORT_##OP));
                                      ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:101:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PU);
     ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/io_mux_reg.h: In function 'PIN_PULLDWN_DIS':
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/esp32/include/esp_assert.h:23:28: error: first argument to '__builtin_choose_expr' not a constant
             _Static_assert(__builtin_choose_expr(__builtin_constant_p(CONDITION), (CONDITION), 1), #MSG);   \
[... cut ...]
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:163:13: note: in expansion of macro 'ASSERT_IF_DPORT_REG'
             ASSERT_IF_DPORT_REG((_r), REG_SET_BIT);                                                                    \
             ^
/Users/jeremy/Code/pycom-micropython-sigfox/pycom-esp-idf/components/soc/esp32/include/soc/io_mux_reg.h:111:5: note: in expansion of macro 'REG_SET_BIT'
     REG_SET_BIT(PIN_NAME, FUN_PD);
     ^
make: *** [build/LOPY_915/release/mods/machtimer_alarm.o] Error 1
```

jeremyherbert, 2017-06-07:
Also fails on Ubuntu 16.04 with the same error.

jeremyherbert, 2017-06-07:
I also note that the travis build is broken

jeremyherbert, 2017-06-07:
It looks like the esp32 compiler listed in the readme might be the problem here. On further inspection of the pycom-esp-idf, I am getting errors about incorrect compiler versions. I'm building the latest compiler now.

jeremyherbert, 2017-06-07:
Ok, looks like I'm almost there. I had to:

1. build mpy-cross as earlier
2. Ignore the toolchain in the readme, it has a bug in the compiler which has since been fixed. Go with this: http://esp-idf.readthedocs.io/en/latest/get-started/linux-setup-scratch.html
3. The correct build command for the bootloader is `make BOARD=LOPY LORA_BAND=USE_BAND_915 TARGET=boot` (readme is missing the `LORA_BAND` part
4. make flash requires pyserial to be installed and access to the usb serial port on the current user. sudo doesn't appear to work

After all of this I was able to build and flash the device. However my LoPy no longer boots. Still working on that one. I just get:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0x00
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:1
load:0x3fff9010,len:12
ho 0 tail 12 room 4
load:0x3fff9020,len:388
load:0x40078000,len:11412
load:0x4009fc00,len:848
entry 0x4009fd9c
```
and then nothing

jeremyherbert, 2017-06-07:
Looks like the bootloader is toast. If I reflash the pycom provided image and then flash my own appimg.bin on top of that, it appears to work

jeremyherbert, 2017-06-07:
FWIW if anyone else is ever struggling with this, the flash command is 
```
python tools/esptool.py write_flash 0x10000 build/LOPY_915/release/appimg.bin
```

run it from the esp32 folder.

yahia13, 2017-06-09:
I had the same error but I followed someone's suggestion in the forum to download a newer toolchain version: https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz . It worked perfectly.

eqrx, 2017-08-02:
I am using the WIPY 2.0 and need to use the newest compiler as @yahia13 pointed out to get my application compiled. After compiling with `make BOARD=WIPY TARGET=app` and  flashing with `make BOARD=WIPY flash` the board hangs at `entry 0x4009fd9c` like the LOPY of @jeremyherbert. Using `python tools/esptool.py write_flash 0x10000 build/WIPY/release/appimg.bin` fixes the problem.

jeremyherbert, 2017-10-24:
For what it's worth, I found out the answer to this. The correct way to flash is:

```
#!/bin/sh
python esptool.py --after no_reset -b 921600 erase_region 0x1000 0x40000
python esptool.py --after no_reset -b 921600 erase_region 0x41000 0x40000
python esptool.py --after no_reset -b 921600 erase_region 0x81000 0x40000
python esptool.py --after no_reset -b 921600 erase_region 0xC1000 0x40000
python esptool.py --after no_reset -b 921600 erase_region 0x101000 0x40000
python esptool.py --after no_reset -b 921600 erase_region 0x141000 0x40000
python esptool.py --after no_reset -b 921600 erase_region 0x181000 0x40000
python esptool.py --after no_reset -b 921600 erase_region 0x1C1000 0x40000

python esptool.py --after no_reset -b 921600 write_flash 0x1000 bootloader.bin
python esptool.py --after no_reset -b 921600 write_flash 0x8000 partitions.bin
python esptool.py --after no_reset -b 921600 write_flash 0x10000 lopy_915.bin
```

`bootloader.bin`, `lopy_915.bin` and `partitions.bin` come from your build directory.

nevercast, 2017-11-30:
Why is it the correct way to erase in blocks of 262k ? Why not use erase_flash ? 

My current command is:
```
esptool.py write_flash -z --flash_mode qio --flash_freq 40m --flash_size 4MB --after hard_reset --chip esp32 --port ${COM} --baud 921600 0x1000 ${BOOT} 0x8000 ${PART} 0x10000 ${APP}
```

This is done after an erase_flash which blanks the entire flash. If there is something ill practice about this I'd like to know about it.

Kind regards,
Josh.

jeremyherbert, 2017-11-30:
I should have been clearer: that is the way that the pycom updater tool does it. I just traced and dumped the commands from there. I am not privy to the reasoning.

nevercast, 2017-12-03:
I did try OTA before trying to flash via USB/serial. Could #81 be related to this issue? (Bootloader not working)

eqrx, 2017-12-04:
@jeremyherbert @yahia13 I just noticed I forgot to thank you for your work. With your information I was able to fix my flash setup. So thank you :)

Xykon, 2017-12-19:
Just FYI the reason we don't erase the entire flash is because that would a) wipe the file system on the device and b) erase the LPWAN MAC address (used by LoPy/SiPy/FiPy) which is also stored in flash.

nevercast, 2017-12-20:
@Xykon right, but you do erase contiguous blocks in multiple chunks, correct?

nevercast, 2018-01-19:
Is this resolved btw? I should mention that I build the firmware every day on a Linux64 ubuntu:xenial machine.