pycom/pycom-micropython-sigfox #21: MIT license violations

dpgeorge, 2017-06-17:
There are some files in this repository which are substantial copies of MIT licensed code from the upstream MicroPython repo and they violate the MIT license because the copyright header is not retained.  The MIT license has only one single requirement, namely: "The above copyright notice and this permission notice shall be included in all copies or substantial portions of the Software."

Please fix this.

The following files at least need fixing:
```
boards/make-pins.py
mpconfigport.h
mods/machpin.c
mods/machspi.c
mods/modmachine.c
mods/modnetwork.h
mods/modnetwork.c
mods/moduos.c
mods/moduselect.c
mods/modusocket.c
mods/modutime.c
```

danicampora, 2017-06-17:
Thanks for ponting this out @dpgeorge.
Fixed here: https://github.com/pycom/pycom-micropython-sigfox/commit/8713fd0b7d749261c18c4659d7cbba0e95808d33

closed 2017-06-17 by danicampora

dpgeorge, 2017-06-17:
Thanks @danicampora for the response, but, sorry, the fix is not correct: you need to go back (in time in the git history) and work out where you copied the files from, then copy the correct license/copyright header.  It definitely shouldn't be "2016 Damien P. George on behalf of Pycom", but rather something like "Copyright (c) 2013, 2014 Damien P. George" with the year depending on which source file it came from.  Also, some files may be copyright other authors.  You will need to go through the history of each file to work this out.

danicampora, 2017-06-17:
@dpgeorge OK, will correct it. 
What to do for files like **boards/make-pins.py** which didn't have a license at all?

danicampora, 2017-06-17:
The files for the ESP32 port where mostly copied from the CC3200 port and some from stmhal. I think I have fixed the files you listed and also a couple others that I noticed myself. https://github.com/pycom/pycom-micropython-sigfox/commit/93ece861e489b9b1d8bdcb9e96eaca5ef9310cd5

dpgeorge, 2017-06-19:
> I think I have fixed the files you listed and also a couple others that I noticed myself. 93ece86

Thanks, that looks a lot better.

> What to do for files like boards/make-pins.py which didn't have a license at all?

In that case it's covered by the general LICENSE file in the root of the repo, so best thing to do would be copy LICENSE to the top of make-pins.py and then add your own stuff to it.