pycom/pycom-micropython-sigfox #29: ESP32: Partitioning inconsistency between pycom.io builds and own builds from the repository

robert-hh, 2017-06-30:
There is obviously an inconsistency in the partition map between firmware builds, made by pycom and those made locally from the github repository. What happens when:
If you have first loaded an pycom.io firmware via updater tool, and then flash a self buitl imgae via "make flash", the device does not start, but just shows a red RGB led. One has to erase the föash completely and reload to get it working.
If you have a self built image working and then load an pycom.io image, the device starts, but the content of the file system is gone.
So: would please someone make these two variants consistent.

quentin-ol, 2017-09-07:
I'm not sure I'm facing the same core issue but the symptoms are identical.
I build the firmware from GitHub sources and flash using the `make BOARD=xxPY [...] flash` command.
No warning/error during compilation ( after installing a newer IDF as#67 suggested), no error when flashing but resetting the device after removing the G23-GND connection ends up with the RGB LED going red.

robert-hh, 2018-06-12:
I assume it is solved meanwhile.

closed 2018-06-12 by robert-hh