pycom/pycom-micropython-sigfox #39: pytrack accelerometer wrong values

ozeta, 2017-07-14:
Hi, I'm playing with the wipy2 + pytrack. 

os.uname()=
(sysname='WiPy', nodename='WiPy', release='1.7.5.b2', version='v1.8.6-694-g25826866 on 2017-06-29', machine='WiPy with E
SP32')

pytrack dfu ver: 0.0.3

I used the turorial script from github to get the accelerometers data in a while loop:

while True:
    print("{} {:>10.3} {:>10.3} {:>10.3} ".format(li.acceleration(), li.pitch(), li.yaw(), li.roll()))
    utime.sleep_ms(500)

I think I got some bugged response

laying on the bench it seems all regular:

(-0.015625, -0.0007324219, 1.014038)      0.884    -0.0414     -0.883
(-0.01708984, 0.001098633, 1.014404)      0.967      0.062     -0.965  

but if I roll the pytrack, both pitch and roll reads some values:

(0.9797363, -0.1525879, -0.07019043)        -86      -8.83       80.3
(1.001099, -0.1734619, -0.1179199)      -83.4      -9.76       78.2 

if I pitch the pytrack, pitch and yaw reads some values:

(-0.03173828, 0.994873, -0.08068848)      -85.4         85      -1.82
(-0.01452637, 0.9941406, -0.07385254)      -85.8       85.7     -0.835

if i yaw the pytrack ( Holding it vertically), it's pitch and yaw again

(0.1431885, -1.013672, 0.03063965)       88.3      -81.8       8.04
(0.1501465, -1.015625, 0.03515625)         88      -81.4        8.4   

but if i turn it 90°, yaw decreases and roll increases:

(-0.9949951, -0.1469727, -0.04418945)      -87.5      -8.39      -81.2  
(-0.9726563, -0.09790039, -0.04040527)      -87.6      -5.74      -83.8 

is that a known bug or I do I have to parse the output data or else?

Bucknalla, 2017-07-16:
Hi @ozeta, apologies for the confusion, I realize that I'd asked you to create an issue on our GitHub repo, but I had no specified to use the pycom/pycom-libraries repository! If you wouldn't mind copy/pasting this to https://github.com/pycom/pycom-libraries/issues as that's where I'm handling the Pytrack/Pysense issues!

I'll close this now so as not to confuse Daniel/the team working on the LoPy/WiPy 2.0/etc. firmware!

Thanks!

ozeta, 2017-07-17:
ops, sure no problem!