pycom/pycom-micropython-sigfox #41: Set iBeacon Advertising Power and Frequency For LoPy

proQreate, 2017-07-19:
Hi,

I've managed to setup the advertisement as such now it functions as an iBeacon. However, I can't find anywhere in the documentation for a way to set the advertising frequency and the actual transmitting power. By Tx Power I do not mean the RSSI which is at the end of the manufacture data.
So I'm wondering if these can be set in some way.

Thanks,

QJ

jmarcelino, 2017-07-20:
For now the advertising frequency is fixed in the C code at https://github.com/pycom/pycom-micropython-sigfox/blob/master/esp32/mods/modbt.c#L202 - currently between adv_int_min (20) to adv_int_max (40ms). 

You could change that and build a custom firmware image, adapt the code so it takes those as parameters or.. wait until someone else does.

However changing the TX power is not supported by Espressif (who make the ESP32 chip) so there is currently no way to do it.

jmarcelino, 2017-08-13:
The latest ESP-IDF includes functions to set transmit (TX) power for BLE. 

esp_ble_tx_power_set(esp_ble_power_type_t power_type, esp_power_level_t power_level)
esp_power_level_t esp_ble_tx_power_get(esp_ble_power_type_t power_type)

in [bt.c](https://github.com/espressif/esp-idf/blob/master/components/bt/bt.c#L423)

It would be great if these could be imported into pycom-esp-idf

danicampora, 2017-08-13:
@jmarcelino we'll do, thanks!

binloan, 2018-04-04:
I want to vote on this issue - Especially on being able to set advertisement intervals.

This functionality is crucial, because if the WiPy/Lopy becomes a peripheral, some older smartphones do not detect the advertisements. This is due to a bad interval time.
I would recommend that the default advertising interval is set to one which is specified in this document by Apple ([https://developer.apple.com/library/content/qa/qa1931/_index.html](https://developer.apple.com/library/content/qa/qa1931/_index.html))

The intervals there work pretty neat with many platforms and different Bluetooth chipsets. Especially on Android where the BT chip quality varies by quiet some degree.