pycom/pycom-micropython-sigfox #51: Visual Studio Code not working

saustin99, 2017-08-18:
Hi. I just got a WiPy 2.0, and discovered that the plugin for Atom is broken (can't find serialport.py), so I'm trying VS Code. I've installed the plugin there, and restarted VSCode. The instruction say to:
     Open 'Global Settings' (this opens automatically after first install).
Where is "Global Settings"? I've been all over the menus, and there's nothing like that.
My first experience with PyCom is definitely not a great start.
Please help!
(I'd enter all that stuff you wnated from the REPL, but I can't get a REPL!)
Thanks,
Steve

robert-hh, 2017-08-18:
Instead of Atom and VS-Code, you can also use a simple serial terminal like Putty on Windows or screen on Mac OS and Linux to get an REPL. Use 115200 Baud, 8 bit, 1 stop bit, no flow control, no parity as settings. Or use a Telnet console for REPL and an FTP client for file transfer. There are many ways to communicate with a WiPy.
For any discussion about how to use the user forum at https://forum.pycom.io/ is most likely the better ṕlace.

danicampora, 2017-08-18:
@saustin99 apologies for this. The plug-ins for ATOM and Visual Studio code were working on the previous versions of both IDE's. The tooling team is working on a fix and new versions of the plug-ins will be released in a few days. In the meantime please use a serial terminal (like putty) and a FTP client (FileZilla) for development.

saustin99, 2017-08-19:
Thanks. But I get nothing. I know I have the right COM port, and it's set just as you say, but nothing appears on the monitor when I reset the WiPy, or hit enter, or anything, for that matter. 

I just wanted to experiment with embedded Python, and this looked good, but maybe I should just return it, and check back in six months or so. I'll see if it's working well for any Windows 10 users. But I would have thought that's a very large percentage of the customer base, so I'm not sure how this is selling. 

In any case, thanks for the quick reply. I appreciate it. 
Steve

robert-hh, 2017-08-19:
Hi Steve, there are many people working with WiPy 2.0 and any OS, including Windows 10. So it's likely a simple connection problem. A few questions:
a) Does the blue heartbeat LED blink every few seconds. If yes, the device is powered and running.
b) Do you see the device's COM port in the Windows device manager? If not, there's a cable or Windows driver issue.
c) If yes, are all jumpers on the Expansion board (if you use one) set? Especially those for RX and TX?

Independently from that you should be able to communicate by telnet, because the device acts by default as AP with the name WIPY_xxxx, password www.pycom.io, the IP 192.168.4.1 and the telnet/ftp user name micro, password python (see also heref https://docs.pycom.io/chapter/toolsandfeatures/repl/telnet.html).

saustin99, 2017-08-19:
Hi, Robert. 

1) Well, the LED blinks every 4 seconds like it said it would, but it's red. Maybe that tells you something? 

2) Yes, COM 4 shows up in the devices window, and disappears if I unplug the board, so I know that's good. And I was able to successfully update the firmware, so at least something can communicate with it. 

3) And yes, the jumpers on the expansion board are all in place. 

Thanks again for the quick reply. 
Steve

robert-hh, 2017-08-19:
I just flashed my LoPy with WiPy code, and the heartbeat is blue. Strange! Are you sure it's an WiPy2?
https://www.pycom.io/product/wipy/

robert-hh, 2017-08-19:
The color should be blue. I have no clue what red means, if it is intentionally. Since I do not have a genuine Wipy here, I loaded a LoPy with the WiPy code, and that shows a blue heartbeat. And there are comments in the forum of people wanting it to change from blue, So I assume that's what it should be. I just looked into the source code: it says blue. So unless @danicampora has another insight (and he may since he is the author of the code), it looks as if the board is defective. You might try another code flash at low speed.

saustin99, 2017-08-19:
Ah. It's not! It says 1.3 on the other side of the board. I just bought it from Adafruit, and assumed they'd send the latest release. Maybe that was dumb. It surprises me that they'd do that (without saying so). They're pretty reputable. 

Does this change anything? As I said, I followed the Getting Started directions on the WiPy site, and can't get either of the IDEs or the board itself to work. (And I don't want to develop in a REPL). 

WiPy doesn't seem to have any different documentation for the 1.3 (in fact, they don't even list it anymore), but the current docs say that the LED is RGB, so maybe they do display red if something's wrong, and blue otherwise. They don't mention the color, just saying that the LED blinks every four seconds if it's ready to go. 

Does the fact that it's an older version give you any clues? 

Thanks for the help! 
Steve 

https://www.pycom.io/product/wipy/

jmarcelino, 2017-08-19:
The 1.3 and 2.0 are completely different products, for the WiPy 1.3 you should follow this documentation 
http://docs.micropython.org/en/latest/wipy/

The Pycom site, this repo and Atom plugin are only about version 2.

Adafruit sells both, the WiPy 2 is at https://www.adafruit.com/product/3338

saustin99, 2017-08-19:
Oh, great. Well Adafruits page for this device had this: 
"Check out the tutorials and quickstart guide at Pycom to see how fast it is to get going in particular, start out with the WiPy 1.0 docs" 

I clicked the first link (had to pick one, so why not?), and they send you right to PyCom. I followed the instructions, which is why I wrote you. 

It turns out Adafruit's policy is that you can only return unopened deliveries, so I likely have to eat this. 

Sigh... Thanks! for your help anyway. Sorry to have dragged you into this. 
Steve 

http://docs.micropython.org/en/latest/wipy/

robert-hh, 2017-08-20:
The WiPy 1.3 is a good device too. It lacks BlueTooth and floating point numbers, but for a first experience with MicroPython is fine. It belongs to the initial branch of MicroPython, developed by Damien George. Out of that, Daniel Campora made a port for the CC3200 chip, which is used by the WiPy 1, and then created his company Pycom with the wider range of devices. The two variants diverged since then, especially in the network and peripherals modules. The language core itself is mostly the same. In my recognition, the pyciom.io branch is faster picking up new features, at the burden of them being not fully tested, while the micropython.org branch only accepts new features for a string reason and after thorough testing.
I was about to ask, whether the device you have is a WiPy 1, since that has a red heartbeat led. Burt I deleted the question after reading, that you had installed the latest firmware.
As far as I recall, the initial mode of WiPy 1.3 offers the REPL by Telnet only. You can switch it to UART, by adding the appropriate statements into boot.py ( see http://docs.micropython.org/en/latest/wipy/wipy/tutorial/repl.html)

saustin99, 2017-08-20:
Hi, Robert. Yes, it has a red heartbeat, and the board says WiPy 1.3 
Your note explains why I didn't get any UART response. But I thought I 
was buying the WiPy2 (I do want the hardware floating point and the 
BlueTooth), so I'll just have to buy another. But not through Adafruit 
again. They won't issue a refund if you've opened the package. Sheesh. I 
can't believe that's ever the case for any customer. Maybe that's the 
point.

Do you happen to know if the expansion board for mine is compatible with 
the WiPy 2.0? I may have to eat that too.

Thanks for sticking with me again. I'm sure I'll enjoy the 2.0 when I 
get it.

Cheers,
Steve

jmarcelino, 2017-08-20:
Which expansion board do you have? This one:
https://www.adafruit.com/product/3344 
with rounded corners and with "Expansionboard V2.1A" on the bottom?
If so you're OK. 

If you have this https://www.adafruit.com/product/2960 with the square corners instead then it probably won't work.

If you're looking for retailers in US check out  Sea Slug Labs, https://seasluglabs.io/collections/wifi/products/wipy 

I'm not related in any way but they're good guys.

saustin99, 2017-08-21:
Cool! Yes, I have the 2.1A. Hah. I'll save that much anyway. Thanks for 
the quick response and the tip on Sea Slug. I'll check them out.

Thanks for all your help,
Steve

sdaniel55, 2018-09-03:
I'll close this issue because it has not had any recent activity. Thank you for your contributions!

closed 2018-09-03 by sdaniel55