pycom/pycom-micropython-sigfox #56: ticks_diff() argument order reversed?

rolandvs, 2017-08-23:
I use both pyboard and xxPy, running both micropython. I used some code _cross-boards_ as far as possible, and I ran into a minor _annoyance_ for a function in `utime` called `ticks_diff()` that really doesn't have to be different at all.

The micropython of Pycom reverses the order of the arguments, whereas the implementation on the original micropython platform uses the IMHO right way to do this. Just to illustrate:

micropython's way
```
# Wait for GPIO pin to be asserted, but at most 500us
start = time.ticks_us()
while pin.value() == 0:
    if time.ticks_diff(time.ticks_us(), start) > 500:
        raise TimeoutError
```

Pycom's micropython way
```
# Wait for GPIO pin to be asserted, but at most 500us
start = time.ticks_us()
while pin.value() == 0:
    if time.ticks_diff(start, time.ticks_us()) > 500:
        raise TimeoutError
```

robert-hh, 2017-12-26:
Actually they are reversed, and that is documented. The other inconsistency is the behaviour in exchanging the arguments to the call. In the MicroPython.org version, simply the sign is changed. In the Pycom version the result is told to be not valid, if the parameters are ticks_diff(new, old) instead of (old, new).
I submitted a PR to make at least this behavior more robust.

husigeza, 2018-08-07:
Hello,
The order of the parameters are swapped in development release 1.9.0.b4 (with uasyncio support), so they are consistent with the MicroPython implemented version of the function. Closing this issue.

closed 2018-08-07 by husigeza