pycom/pycom-micropython-sigfox #57: LoRaWAN AddChannel Bug

timrid, 2017-08-25:
(sysname='LoPy', nodename='LoPy', release='1.7.9.b3', version='v1.8.6-747-gc4ec6110 on 2017-08-09', machine='LoPy with ESP32', lorawan='1.0.0')

I tried to Add the FSK Channel from LoRaWAN 868 MHz (DR = 7) with the following command:

`lora.add_channel(8, frequency=868800000, dr_min=7, dr_max=7)`

An then i get the Error:

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ValueError: invalid argument(s) value
```

I think the Error is in: 

```
esp32/mods/modlora.c :
lora_validate_data_rate()
```
It aborts, when the datarate is greater than DR_6 and that is wrong, because in 868 MHz band it can go to DR_7.

jmarcelino, 2017-08-25:
It's not really a bug but simply reflects that FSK has not been implemented in this port of LoRaMAC. At the very least code to handle DR 7 would need to be added.

timrid, 2017-08-26:
And is it planed to implement FSK in the LoRaMAC in the near future?

danicampora, 2017-08-26:
@timrid yes we will support it somewhere in Q4 this year.

robertkeizer, 2018-12-12:
Hey @danicampora what's the timeline on this?