pycom/pycom-micropython-sigfox #58: Fault in unix version

henriqueng, 2017-08-25:
Please include the following information when submitting a bug report:

* System: Ubuntu 16.04 LTS x64

* Environment setup:
    1. git clone https://github.com/pycom/pycom-micropython-sigfox.git
    2. cd pycom-micropython-sigfox/unix
    3. make axtls
    4. make

* Micropython test:
```   ./micropython
   MicroPython ccb10e3 on 2017-08-25; linux version
   Use Ctrl-D to exit, Ctrl-E for paste mode
   >>> print('Hello World')
   Segmentation fault (core dumped)
```

* make test:
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
cd ../tests && MICROPY_MICROPYTHON=../unix/micropython ./run-tests
FAIL  basics/0prelim.py
pass  basics/andor.py
FAIL  basics/array1.py
FAIL  basics/array_add.py
FAIL  basics/array_construct.py
FAIL  basics/array_construct2.py
skip  basics/array_construct_endian.py
FAIL  basics/array_q.py
FAIL  basics/assign1.py
FAIL  basics/async_await.py
FAIL  basics/async_await2.py
FAIL  basics/async_def.py
FAIL  basics/async_for.py
FAIL  basics/async_for2.py
FAIL  basics/async_with.py
FAIL  basics/async_with2.py
FAIL  basics/attrtuple1.py
FAIL  basics/bool1.py
FAIL  basics/boundmeth1.py
FAIL  basics/break.py
FAIL  basics/builtin_abs.py
FAIL  basics/builtin_allany.py
FAIL  basics/builtin_bin.py
FAIL  basics/builtin_callable.py
FAIL  basics/builtin_chr.py
FAIL  basics/builtin_compile.py
FAIL  basics/builtin_delattr.py
FAIL  basics/builtin_dir.py
FAIL  basics/builtin_divmod.py
pass  basics/builtin_ellipsis.py
FAIL  basics/builtin_eval.py
FAIL  basics/builtin_eval_error.py
FAIL  basics/builtin_hash.py
FAIL  basics/builtin_hex.py
FAIL  basics/builtin_id.py
FAIL  basics/builtin_issubclass.py
FAIL  basics/builtin_len1.py
FAIL  basics/builtin_minmax.py
FAIL  basics/builtin_oct.py
FAIL  basics/builtin_ord.py
FAIL  basics/builtin_override.py
FAIL  basics/builtin_pow.py
FAIL  basics/builtin_property.py
FAIL  basics/builtin_range.py
FAIL  basics/builtin_reversed.py
FAIL  basics/builtin_round.py
FAIL  basics/builtin_slice.py
FAIL  basics/builtin_sorted.py
FAIL  basics/builtin_sum.py
[... 515 lines cut ...]
FAIL  stress/list_sort.py
FAIL  unicode/file1.py
FAIL  unicode/file2.py
pass  unicode/unicode.py
pass  unicode/unicode_chr.py
FAIL  unicode/unicode_id.py
FAIL  unicode/unicode_index.py
FAIL  unicode/unicode_iter.py
FAIL  unicode/unicode_ord.py
FAIL  unicode/unicode_pos.py
FAIL  unicode/unicode_str_format.py
FAIL  unicode/unicode_str_modulo.py
FAIL  unicode/unicode_subscr.py
FAIL  unix/extra_coverage.py
FAIL  unix/ffi_callback.py
FAIL  unix/ffi_float.py
FAIL  unix/ffi_float2.py
518 tests performed (17778 individual testcases)
38 tests passed
32 tests skipped: array_construct_endian bytearray_construct_endian bytes_add_endian bytes_construct_endian repl_emacs_keys vfs_fat_fileio vfs_fat_fsusermount vfs_fat_oldproto vfs_fat_ramdisk complex1 int_big_float true_value types native_closure native_const native_misc viper_addr viper_args viper_binop_arith viper_binop_comp viper_binop_comp_imm viper_binop_multi_comp viper_cond viper_error viper_misc viper_ptr16_load viper_ptr16_store viper_ptr32_load viper_ptr32_store viper_ptr8_load viper_ptr8_store viper_subscr
480 tests failed: 0prelim array1 array_add array_construct array_construct2 array_q assign1 async_await async_await2 async_def async_for async_for2 async_with async_with2 attrtuple1 bool1 boundmeth1 break builtin_abs builtin_allany builtin_bin builtin_callable builtin_chr builtin_compile builtin_delattr builtin_dir builtin_divmod builtin_eval builtin_eval_error builtin_hash builtin_hex builtin_id builtin_issubclass builtin_len1 builtin_minmax builtin_oct builtin_ord builtin_override builtin_pow builtin_property builtin_range builtin_reversed builtin_round builtin_slice builtin_sorted builtin_sum builtin_type bytearray1 bytearray_add bytearray_append bytearray_construct bytearray_longint bytearray_slice_assign bytes bytes_add bytes_compare2 bytes_compare3 bytes_construct bytes_count bytes_find bytes_format_modulo bytes_gen bytes_large bytes_mult bytes_partition bytes_replace bytes_split bytes_strip bytes_subscr class1 class2 class3 class_bind_self class_binop class_call class_contains class_descriptor class_emptybases class_getattr class_inherit1 class_inherit_mul class_instance_override class_item class_new class_number class_staticclassmethod class_store class_store_class class_str class_super class_super_object class_use_other closure1 closure2 closure_defargs closure_manyvars closure_namedarg compare_multi comprehension1 containment continue decorator del_attr del_deref del_global del_local del_name del_subscr dict2 dict_clear dict_construct dict_copy dict_del dict_from_iter dict_fromkeys dict_get dict_intern dict_iterator dict_pop dict_popitem dict_setdefault dict_specialmeth dict_update dict_views enumerate equal equal_class errno1 except_match_tuple exception1 exception_chain exceptpoly exceptpoly2 exec1 filter floordivide for1 for2 for3 for_break for_range for_return frozenset1 frozenset_add frozenset_copy frozenset_set fun1 fun2 fun3 fun_annotations fun_calldblstar fun_calldblstar2 fun_calldblstar3 fun_callstar fun_callstardblstar fun_defargs fun_defargs2 fun_error fun_kwargs fun_kwonly fun_kwvarargs fun_largestate fun_name fun_str fun_varargs gc1 gen_yield_from gen_yield_from_close gen_yield_from_ducktype gen_yield_from_exc gen_yield_from_iter gen_yield_from_send gen_yield_from_stopped gen_yield_from_throw generator1 generator2 generator_args generator_close generator_closure generator_exc generator_send getattr getattr1 getitem globals_del hasattr1 ifcond int1 int2 int_big_and int_big_and3 int_big_cmp int_big_div int_big_error int_big_lshift int_big_mod int_big_mul int_big_or int_big_or2 int_big_or3 int_big_pow int_big_rshift int_big_unary int_big_xor int_big_xor2 int_big_xor3 int_big_zeroone int_bytes int_constfolding int_divmod int_divzero int_long int_mpz int_small is_isnot iter0 iter1 iter2 iter_of_iter lambda1 lambda_defargs lexer list1 list_clear list_compare list_copy list_count list_extend list_index list_insert list_mult list_pop list_remove list_reverse list_slice list_slice_3arg list_slice_assign list_slice_assign_grow list_sort list_sum map memoryerror memoryview1 memoryview2 memoryview_gc module1 module2 namedtuple1 object1 object_dict object_new op_error parser print python34 return1 scope seq_unpack set1 set_add set_clear set_copy set_difference set_discard set_intersection set_isdisjoint set_isfooset set_iter set_pop set_specialmeth set_symmetric_difference set_union set_unop set_update setattr1 slice_attrs slice_bignum slots_bool_len special_methods string1 string_center string_compare string_count string_cr_conversion string_crlf_conversion string_endswith string_escape string_find string_format string_format2 string_format_error string_format_modulo string_index string_join string_large string_mult string_partition string_replace string_repr string_rfind string_rindex string_rpartition string_rsplit string_slice string_split string_splitlines string_startswith string_strip struct1 struct2 subclass_classmethod subclass_native1 subclass_native2_list subclass_native2_tuple subclass_native3 subclass_native4 subclass_native5 subclass_native_buffer subclass_native_cmp subclass_native_specmeth syntaxerror try1 try2 try3 try4 try_continue try_finally2 try_finally_return try_finally_return2 tuple1 tuple_compare tuple_count tuple_index tuple_mult types1 types2 unary_op unboundlocal unpack1 while1 while_cond while_nest_exc with1 with_break with_continue with_return zip cmd_optimise cmd_showbc cmd_verbose repl_basic repl_cont btree1 framebuf1 machine_pinbase machine_pulse ticks_diff time_ms_us ubinascii_a2b_base64 ubinascii_b2a_base64 ubinascii_hexlify ubinascii_unhexlify uctypes_array_assign_le uctypes_array_assign_native_le uctypes_bytearray uctypes_le uctypes_le_float uctypes_native_float uctypes_native_le uctypes_ptr_le uctypes_ptr_native_le uctypes_sizeof uctypes_sizeof_native uheapq1 ujson_dumps ujson_dumps_extra ujson_dumps_float ujson_load ujson_loads ujson_loads_float urandom_basic urandom_extra ure1 ure_error ure_group ure_namedclass ure_split ure_split_empty uzlib_decompress array_construct builtin_float_minmax bytearray_construct bytes_construct cmath_fun_special float1 float2int_doubleprec float_array float_divmod float_divmod_relaxed float_struct int_divzero int_power list_index math_fun math_fun_special string_format string_format2 string_format_fp30 string_format_modulo string_format_modulo2 gen_context gen_context2 import [... cut ...] error decorator decorator_error emg_exc heap_lock heapalloc meminfo memstats opt_level features non_compliant non_compliant_lexer print_exception recursion recursive_data recursive_iternext rge_sm sys_exc_info dict_copy dict_create list_sort file1 file2 unicode_id unicode_index unicode_iter unicode_ord unicode_pos unicode_str_format unicode_str_modulo unicode_subscr extra_coverage ffi_callback ffi_float ffi_float2
Makefile:190: recipe for target 'test' failed
make: *** [test] Error 1

Thank you!

robert-hh, 2017-08-25:
if you need the Unix version, better go to www.micropython.org and the repository https://github.com/micropython/micropython.

henriqueng, 2017-08-25:
Ok. I intented to port a LWM2M (and CoAP also) library to pycom-micropython. Unfortunately, my pycom board was not delivered yet :(