pycom/pycom-micropython-sigfox #63: ustruct Guru Meditation Error crash and blocks the device

syndycat, 2017-08-31:
LopY 1.7.8.b1

This type of crash  blocks the device. 

Simple to reproduce. Type in console:

a= 0xaabbccdd
a.to_bytes(4)
b'\xdd\xcc\xbb\xaa'

import ustruct
ustruct.pack('4b',a)

Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400ee210  PS      : 0x00060030  A0      : 0x800ee403  A1      : 0x3ffc6830  
A2      : 0x3ffc6af0  A3      : 0x3ffc6841  A4      : 0x00000001  A5      : 0x3ffe83f2  
A6      : 0x000000fb  A7      : 0xffffffff  A8      : 0x00000000  A9      : 0x0000ccdd  
A10     : 0x00000002  A11     : 0x3ffe83f0  A12     : 0x0000ccdd  A13     : 0x3ffe83ef  
A14     : 0x00000010  A15     : 0x3ffe83f1  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000004  LBEG    : 0x4000c28c  LEND    : 0x4000c296  LCOUNT  : 0x00000000  

Backtrace: 0x400ee210:0x3ffc6830 0x400ee400:0x3ffc6850 0x400f5ef0:0x3ffc6870 0x400f77ae:0x3ffc68b0 0x400f7801:0x3ffc68f0 0x400f078b:0x3ffc6920 0x400ecc5d:0x3ffc6950 0x400eccc5:0x3ffc6970 0x400f883d:0x3ffc6990 0x400f0710:0x3ffc6a30 0x400ecc5d:0x3ffc6ab0 0x400ecc8a:0x3ffc6ad0 0x400d9953:0x3ffc6af0 0x400d9bf4:0x3ffc6b90 0x400d8c28:0x3ffc6bd0

Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4008a199  PS      : 0x00060033  A0      : 0x80088fd6  A1      : 0x3ffc6470  
A2      : 0x3ffc64c0  A3      : 0x3ffc6490  A4      : 0x00000000  A5      : 0x3ffc483c  
A6      : 0x00000000  A7      : 0x3ffc4858  A8      : 0x00000012  A9      : 0x3ffc6440  
A10     : 0x3ffc6508  A11     : 0x3ffc6490  A12     : 0x3ffb798c  A13     : 0x3ffc6850  
A14     : 0x3ffc6af0  A15     : 0x00000000  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000  

Backtrace: 0x4008a199:0x3ffc6470 0x40088fd3:0x3ffc6490 0x4008b2f5:0x3ffc64c0 0x4008b66a:0x3ffc6690 0x4008af75:0x3ffc66d0 0x4008b0f0:0x3ffc6750 0x4008168a:0x3ffc6770 0x400ee20d:0x3ffc6830 0x400ee20d:0x3ffc6850 0x400f5ef0:0x3ffc6870 0x400f77ae:0x3ffc68b0 0x400f7801:0x3ffc68f0 0x400f078b:0x3ffc6920 0x400ecc5d:0x3ffc6950 0x400eccc5:0x3ffc6970 0x400f883d:0x3ffc6990 0x400f0710:0x3ffc6a30 0x400ecc5d:0x3ffc6ab0 0x400ecc8a:0x3ffc6ad0 0x400d9953:0x3ffc6af0 0x400d9bf4:0x3ffc6b90 0x400d8c28:0x3ffc6bd0

robert-hh, 2017-09-01:
There were a series of commits in the main branch of micropython.org in that area last night.
https://github.com/micropython/micropython/commits/master
Especially commits:
https://github.com/micropython/micropython/commit/2daacc5ceef7e36ac3c88b8946088b6b3513993b
https://github.com/micropython/micropython/commit/79d5acbd01fed21e1c8e46b06690072375de0bfe
https://github.com/micropython/micropython/commit/793d826d9df67c3f544505beda63ca36d8dfa1c4

robert-hh, 2017-09-01:
Using the file py/modstruct.c from the micropython.org branch fixes that problem.

danicampora, 2017-09-03:
This is being fixed with @robert-hh's pulls request, but please note that doing:

```python
a = 0xaabbccdd
a.to_bytes(4)
```

it wrong because the `to_bytes()` method requires the byte order to be passed as well (https://docs.python.org/3.2/library/stdtypes.html#int.to_bytes), so allowing it to work without that parameter was wrong. From the next release onwards you will have to do:

```python
a.to_bytes(4, 'little')
```

Cheers,
Daniel

closed 2018-09-03 by sdaniel55