pycom/pycom-micropython-sigfox #71: Pycom Firmware Update not working

Jan-Valdres, 2017-09-18:
Using LoPY1.0r with Expansion board. Trying to use Pycom Firmware Update.app version 1.1.1.b1 running on a MacBook Pro with macOS Sierra version 10.12.4.

Getting message:
-----------------------------------------------------------------------------
The upgrade failed, please check the connections and try the steps again.

This is the log of the upgrading process

Connecting...........................................

Failure!
-----------------------------------------------------------------------------

I've checked and rechecked. A colleague has tested the board with a PC and everything seemed
OK, but it won't work on my Mac.

jmarcelino, 2017-09-18:
Did you select the select correct port? Which one did you pick?

Jan-Valdres, 2017-09-18:
I only get to choose from two:
/dev/cu.usbserial-DQ00DBKM (the default one)
and
/dev/cu.Bluetuooth-Incoming-Port

I used the first one for most of my trials. (In desperation I tried the second one once or twice with the same result.

jmarcelino, 2017-09-18:
Thanks. I also noticed you are have an old version of the updater, version 1.1.2.b1 became available 2 months ago, and 1.1.1b1 doesn’t work anymore 

https://forum.pycom.io/topic/1535/important-new-firmware-update-tool-released-version-1-1-2-b1

Jan-Valdres, 2017-09-19:
Thank you for quick reply. I open the page you refer to. That page refers me to https://www.pycom.io/downloads/ where I click on "Pycom upgrader macOS (10.11 or higher)". Then I obtain Pycom_firmware_update_1.1.2.b1.dmg which I run on my mac (after having uninstalled the previous version). After having completed the installation I hav a Pycom Firmware update.app which was made Wednesday 8 February 2017 18:22 and is designated Version 1.1.1b1, and the problem remains. Maybe there is something wrong with the .dmg-file?

danicampora, 2017-09-19:
@Jan-Valdres are you putting the device into programming mode as explained by the tool before starting?

Jan-Valdres, 2017-09-19:
If putting the device into programming mode means connecting pin G23 to GND, yes. (That's the only modification the instructions tell me to do.)

robert-hh, 2017-09-19:
When you connect the LoPy  with a serial terminal like screen at 115200 baud, pull G23 to GND and hit reset at Lopy, what do you get on the terminal screen?

Jan-Valdres, 2017-09-19:
I'm sorry, but I'm just a beginner. I wouldn't know how to do just that ...

robert-hh, 2017-09-19:
Did you already connect to you LoPy, e.g. by PyMakr or the screen program? or is Updating the first thing you try?

Jan-Valdres, 2017-09-19:
Yes. I start with the updating because that's recommended by the supplier.

robert-hh, 2017-09-20:
The device should already have a firmware loaded, which you can use to verify the basic communication. Try to connect it to your PC, using a serial terminal emulation program like screen in the Mac terminal window. Connect the LoPy to your mac, open the Mac Terminal and enter the command 
"screen /dev/cu.usbserial-DQ00DBKM 115200"
If all is good, you should get the Python prompt. >>>
Connect then G23 to GND and push the reset button on the LoPy. This button is next to the RGB led, which should show a blue heartbeat signal (a short blink every 4 seconds)
Then you should see a few lines, ending in "waiting for upload".
That tells you, that the connection to the LoPy is fine. At that point you can close the mac terminal window and start the pycom updater. If you see a different message, just let us know.

tho068, 2018-02-08:
I have the same problem. Using Arch Linux. I however, see the correct output from screen, but the updater is not able to connect.

asergios, 2018-02-27:
@tho068

Same. Have you been able to fix it?

jmarcelino, 2018-02-27:
@asergios 
These problems on Linux are usually either bad USB cables or device permissions problems

asergios, 2018-02-27:
@jmarcelino

You are right, I was running the updater with sudo so I thought that wouldnt be a problem. But adding rwx permissions to /dev/ttyUSB0 fixed the problem.

Thanks!

tho068, 2018-02-28:
I fixed the issue by installing all the python dependencies. I think that the error msgs should be improved in the updater. 

Saying not able to connect, without giving any reason is hard to debug.

tho068, 2018-02-28:
Also, giving a list of deps more accesible in the docs might be a good idea

sdaniel55, 2018-09-03:
I'll close this issue because it has not had any recent activity. Thank you for your contributions!

closed 2018-09-03 by sdaniel55