pycom/pycom-micropython-sigfox #72: Atom editor crashes when trying to connect to Pycom

lesure1, 2017-09-26:
Hello,

I am trying to connect to Pycom (1.1.3) from Atom editor (Atom: 1.20.1, Electron: 1.6.9, Chrome: 56.0.2924.87, Node: 7.4.0) on Ubuntu 14.04. I added Pymkr 1.1.3 automatically using the Atom GUI. The Atom editor crashes when clicking on connect, after retrieving and setting the USB serial port for Pycom. The message displayed is "Editor has crashed. Please report this issue". I am totally new to the environment and trying to do my first set up. Could anyone help me identify the source of the problem and fix it?

Thank you!

robert-hh, 2017-09-26:
The questions around pymakr & atom are discussed here in detail: https://forum.pycom.io/topic/1026/pymakr-atom-plugin-released/68

lesure1, 2017-09-26:
Okay, thanks. I will post it there.

sdaniel55, 2018-09-03:
I'll close this issue because it has not had any recent activity. Thank you for your contributions!

closed 2018-09-03 by sdaniel55