pycom/pycom-micropython-sigfox #73: pytrack GPS coordinate give a failure to lora downlink message

ninnux, 2017-09-27:
My System:
(sysname='LoPy', nodename='LoPy', release='1.8.0.b1', version='v1.8.6-760-g90b72952 on 2017-09-01', machine='LoPy with ESP32', lorawan='1.0.0')
pytrack fw 0.0.4

My fault main.py:

""" OTAA Node example compatible with the LoPy Nano Gateway """

from network import LoRa
import socket
import binascii
import struct
import time
import config
import time
import gc
from machine import Timer
from L76GNSS import L76GNSS
from pytrack import Pytrack
from network import LoRa
import pycom

# initialize LoRa in LORAWAN mode.
lora = LoRa(mode=LoRa.LORAWAN)

# create an OTA authentication params
dev_eui = binascii.unhexlify('00 00 00 00 00 00 00 05'.replace(' ',''))
app_eui = binascii.unhexlify('00 00 00 00 00 00 00 05'.replace(' ',''))
app_key = binascii.unhexlify('AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA AA'.replace(' ',''))

# set the 3 default channels to the same frequency (must be before sending the OTAA join request)
lora.add_channel(0, frequency=config.LORA_FREQUENCY, dr_min=0, dr_max=5)
lora.add_channel(1, frequency=config.LORA_FREQUENCY, dr_min=0, dr_max=5)
lora.add_channel(2, frequency=config.LORA_FREQUENCY, dr_min=0, dr_max=5)

# join a network using OTAA
lora.join(activation=LoRa.OTAA, auth=(dev_eui, app_eui, app_key), timeout=0, dr=config.LORA_NODE_DR)

#pycom.heartbeat(False)

# wait until the module has joined the network
while not lora.has_joined():
    time.sleep(2.5)
    print('Not joined yet...')
print('Joined now')

# remove all the non-default channels
for i in range(3, 16):
    lora.remove_channel(i)

# create a LoRa socket
s = socket.socket(socket.AF_LORA, socket.SOCK_RAW)

# set the LoRaWAN data rate
s.setsockopt(socket.SOL_LORA, socket.SO_DR, config.LORA_NODE_DR)

# make the socket blocking
s.setblocking(False)

time.sleep(5.0)

## Enable garbage collection

time.sleep(2)
gc.enable()

## Set up GPS with modest timeout

py = Pytrack()
l76 = L76GNSS(py, timeout=30)

chrono = Timer.Chrono()
chrono.start()
#ledstatus=0x7f7f00 # giallo 

while True:
    coord = l76.coordinates()  ## Gets coordinates
    if not coord == (None, None):
        lat, lon = coord
        print(lat, lon)
    else:
        print("No coordinates found")
   	lat="None"
	lon="None" 
    s.send(b''+str(lat)+','+str(lon))  ## Send last known coordinates
    print("messge sent")
    time.sleep(4)
    rx, port = s.recvfrom(256)
    if rx:
        print('Received: {}, on port: {}'.format(rx, port))
    gc.mem_free()
    time.sleep(6)

If I comment "coord = l76.coordinates()" line node receive downlink message, otherwise messages remain in server queue and never arrive to node.

danicampora, 2017-09-27:
Hello, the call to get the coordinates is blocking everything until the GPS gets a valid fix. You can try using a different thread for the LoRa part while you wait for the GPS.

ninnux, 2017-09-27:
OK, thank you. I will try using thread for LoRa part. 
Anyway I noted this issue with GPS fix too.