pycom/pycom-micropython-sigfox #76: Making mpy-cross  is failing  1.9.0

syndycat, 2017-10-05:
1. I did a with Fresh new git pull and submodules update
     pycom-esp-idf.git
     pycom-micropython-sigfox 

Then: 
                    **$ cd mpy-cross
                    $ make all**
                    Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
                    ../py/py.mk:271: warning: overriding recipe for target 'make'
                    ../py/py.mk:259: warning: ignoring old recipe for target 'make'
                    ../py/py.mk:271: warning: overriding recipe for target 'V=1'
                    ../py/py.mk:259: warning: ignoring old recipe for target 'V=1'
                    ../py/py.mk:271: warning: overriding recipe for target 'BOARD=LOPY'
                    ../py/py.mk:259: warning: ignoring old recipe for target 'BOARD=LOPY'
                    ../py/py.mk:271: warning: overriding recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:259: warning: ignoring old recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:271: warning: overriding recipe for target 'flash'
                    ../py/py.mk:259: warning: ignoring old recipe for target 'flash'
                    ../py/py.mk:279: warning: overriding recipe for target 'make'
                    ../py/py.mk:271: warning: ignoring old recipe for target 'make'
                    ../py/py.mk:279: warning: overriding recipe for target 'V=1'
                    ../py/py.mk:271: warning: ignoring old recipe for target 'V=1'
                    ../py/py.mk:279: warning: overriding recipe for target 'BOARD=LOPY'
                    ../py/py.mk:271: warning: ignoring old recipe for target 'BOARD=LOPY'
                    ../py/py.mk:279: warning: overriding recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:271: warning: ignoring old recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:279: warning: overriding recipe for target 'flash'
                    ../py/py.mk:271: warning: ignoring old recipe for target 'flash'
                    ../py/py.mk:283: warning: overriding recipe for target 'make'
                    ../py/py.mk:279: warning: ignoring old recipe for target 'make'
                    ../py/py.mk:283: warning: overriding recipe for target 'V=1'
                    ../py/py.mk:279: warning: ignoring old recipe for target 'V=1'
                    ../py/py.mk:283: warning: overriding recipe for target 'BOARD=LOPY'
                    ../py/py.mk:279: warning: ignoring old recipe for target 'BOARD=LOPY'
                    ../py/py.mk:283: warning: overriding recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:279: warning: ignoring old recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:283: warning: overriding recipe for target 'flash'
                    ../py/py.mk:279: warning: ignoring old recipe for target 'flash'
                    ../py/py.mk:287: warning: overriding recipe for target 'make'
                    ../py/py.mk:283: warning: ignoring old recipe for target 'make'
                    ../py/py.mk:287: warning: overriding recipe for target 'V=1'
                    ../py/py.mk:283: warning: ignoring old recipe for target 'V=1'
                    ../py/py.mk:287: warning: overriding recipe for target 'BOARD=LOPY'
                    ../py/py.mk:283: warning: ignoring old recipe for target 'BOARD=LOPY'
                    ../py/py.mk:287: warning: overriding recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:283: warning: ignoring old recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:287: warning: overriding recipe for target 'flash'
                    ../py/py.mk:283: warning: ignoring old recipe for target 'flash'
                    ../py/py.mk:291: warning: overriding recipe for target 'make'
                    ../py/py.mk:287: warning: ignoring old recipe for target 'make'
                    ../py/py.mk:291: warning: overriding recipe for target 'V=1'
                    ../py/py.mk:287: warning: ignoring old recipe for target 'V=1'
                    ../py/py.mk:291: warning: overriding recipe for target 'BOARD=LOPY'
                    ../py/py.mk:287: warning: ignoring old recipe for target 'BOARD=LOPY'
                    ../py/py.mk:291: warning: overriding recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:287: warning: ignoring old recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:291: warning: overriding recipe for target 'flash'
                    ../py/py.mk:287: warning: ignoring old recipe for target 'flash'
                    ../py/mkrules.mk:25: warning: overriding recipe for target 'make'
                    ../py/py.mk:291: warning: ignoring old recipe for target 'make'
                    ../py/mkrules.mk:25: warning: overriding recipe for target 'V=1'
                    ../py/py.mk:291: warning: ignoring old recipe for target 'V=1'
                    ../py/mkrules.mk:25: warning: overriding recipe for target 'BOARD=LOPY'
                    ../py/py.mk:291: warning: ignoring old recipe for target 'BOARD=LOPY'
                    ../py/mkrules.mk:25: warning: overriding recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/py.mk:291: warning: ignoring old recipe for target 'LORA_BAND=USE_BAND_868'
                    ../py/mkrules.mk:25: warning: overriding recipe for target 'flash'
                    ../py/py.mk:291: warning: ignoring old recipe for target 'flash'
                    ../py/mkrules.mk:24: *** mixed implicit and normal rules: deprecated syntax
                    ../py/mkrules.mk:30: warning: overriding recipe for target 'make'
[... 93 lines cut ...]
                    make: Circular ESPPORT/genhdr/qstr.i.last <- V=1 dependency dropped.
                    make: Circular ESPPORT/genhdr/qstr.i.last <- BOARD=LOPY dependency dropped.
                    make: Circular ESPPORT/genhdr/qstr.i.last <- LORA_BAND=USE_BAND_868 dependency dropped.
                    make: Circular ESPPORT/genhdr/qstr.i.last <- flash dependency dropped.
                    make: Circular ESPPORT/genhdr/mpversion.h <- make dependency dropped.
                    make: Circular ESPPORT/genhdr/mpversion.h <- V=1 dependency dropped.
                    make: Circular ESPPORT/genhdr/mpversion.h <- BOARD=LOPY dependency dropped.
                    make: Circular ESPPORT/genhdr/mpversion.h <- LORA_BAND=USE_BAND_868 dependency dropped.
                    make: Circular ESPPORT/genhdr/mpversion.h <- flash dependency dropped.
                    GEN ESPPORT/genhdr/qstr.i.last
                    gcc: error: V=1: No such file or directory
                    gcc: error: BOARD=LOPY: No such file or directory
                    gcc: error: LORA_BAND=USE_BAND_868: No such file or directory
                    gcc: error: flash: No such file or directory
                    gcc: error: ESPPORT/tmp: No such file or directory
                    gcc: error: V=1: No such file or directory
                    gcc: error: BOARD=LOPY: No such file or directory
                    gcc: error: LORA_BAND=USE_BAND_868: No such file or directory
                    gcc: error: flash: No such file or directory
                    gcc: error: V=1: No such file or directory
                    gcc: error: BOARD=LOPY: No such file or directory
                    gcc: error: LORA_BAND=USE_BAND_868: No such file or directory
                    gcc: error: flash: No such file or directory
                    gcc: error: ESPPORT/genhdr/qstr.i.last: No such file or directory
                    make: *** No rule to make target 'ESPPORT/genhdr/qstr.i.last', needed by 'ESPPORT/genhdr/qstr.split'.  Stop.

Xykon, 2017-10-06:
Hello, please see this link for a solution: https://github.com/micropython/micropython/issues/2298

I will see if we can get this fixed in the next release.

Xykon, 2017-10-10:
This should now be fixed. Can you please check to confirm?

sdaniel55, 2018-09-03:
Hi @syndycat 👋 
I'll close this issue because it has not had any recent activity and should be resolved. Thank you for your contributions!

closed 2018-09-03 by sdaniel55