pycom/pycom-micropython-sigfox #77: SiPy Firmware Updater not working

fxwalsh, 2017-10-05:
Hi
I'm using SiPy board with Expansion Board 2.
Details as follows:
(sysname='SiPy', nodename='SiPy', release='1.6.3.b2', version='v1.8.6-470-gf41d0f27 on 2017-02-22', machine='SiPy with ESP32', sigfox='1.0.1')

I followed the instruction for updating the firmware.
All progressed as expected (selected siPy board  and Europe region)  until the downloading... part where it just stalls. 
[image]

The next button is available but if I click it just says Result.
I would expect to get the ID and PAC.
I'm using Upgrade tool on Windows10 (v1.1.2.b1)

I have managed to connect and run code on the board using Atom and relevant plugin - ID and PAC are all 'fffff's. 
My guess is the Firmware Updater is not working

Any help would be greatly appreciated

Thank you!

danicampora, 2017-10-06:
@fxwalsh Sorry about this. The issue is that we ran out of the Sigfox IDs for the Europe devices. We are now releasing more IDs. It should be fixed in a few hours. Apologies for the inconvenience.

danicampora, 2017-10-06:
@fxwalsh problem solved. Please try updating you SiPy again. It should work now. Sorry for the issues.

fxwalsh, 2017-10-09:
@danicampora That's sorted it. Thanks!

closed 2017-10-09 by fxwalsh