pycom/pycom-micropython-sigfox #78: Build fails when creating wipy.bin from esptool.py

peterson79, 2017-10-06:
Hi, 

I am trying to build WiPy 2.0 from source on OS X.  I initially ran into issues with the xtensa gcc compiler downloaded from the readme link, so now I am using the most recent version from the Espressif website.  Everything builds and links but the last step of creating the binary fails:

```
xtensa-esp32-elf-size build/WIPY/release/application.elf
   text	   data	    bss	    dec	    hex	filename
 995078	 200788	  57080	1252946	 131e52	build/WIPY/release/application.elf
Building partitions from lib/partitions.csv...
python /Users/jasonpeterson/Documents/pycom/pycom-esp-idf/components/partition_table/gen_esp32part.py -q lib/partitions.csv build/WIPY/release/lib/partitions.bin
IMAGE build/WIPY/release/wipy.bin
python /Users/jasonpeterson/Documents/pycom/pycom-esp-idf/components/esptool_py/esptool/esptool.py --chip esp32 elf2image --flash_mode dio --flash_freq 40m -o build/WIPY/release/wipy.bin build/WIPY/release/application.elf
esptool.py v2.0.1

A fatal error occurred: Segment loaded at 0x401ac9d8 lands in same 64KB flash mapping as segment loaded at 0x401ac9c0. Can't generate binary. Suggest changing linker script or ELF to merge sections.
make: *** [build/WIPY/release/wipy.bin] Error 2
make: *** Deleting file `build/WIPY/release/wipy.bin'
```

1. I downloaded the repo and followed the instructions to build for esp32
2. Ran into compile issues due to bad xtensa gcc compiler
3. Downloaded and installed the current xtensa gcc compiler 
4. make V=1 BOARD=WIPY TARGET=app clean
5. make V=1 BOARD=WIPY TARGET=app
6. ERROR from esptool.py (see the code block above)

I expected a wipy.bin.

What I got was the error above :)

Any insight appreciated! Thanks!

robert-hh, 2017-10-07:
Do you have many Python scripts in your frozen module? It looks like the image gets too large.

peterson79, 2017-10-07:
Yeah, I thought it was a size problem also.  However I don't have enough experience with the code to know where to start.

In the "frozen" folder there is only "_boot.py"

peterson79, 2017-10-07:
Sorry, hit the wrong button!

Are there other locations for "frozen modules?  I am using the default source tree, the only change I have made is updating the compiler.

peterson79, 2017-10-07:
Did your test build finish with a binary?  Or mean it was 12k larger then mine?

robert-hh, 2017-10-07:
Oh yes, the finished build is then loaded to the WiPy with
make flash
That's my standard procedure. I do not use the pycom updater. A full sequence of make commands for WIPY is:
make clean
make TARGET=boot
make
make flash

danicampora, 2017-10-07:
@peterson79 unfortunately the new compiler doesn't work yet. This is a known issue. In order to make the new compiler work we need the new IDF, but the new IDF has a bunch of issues that prevent using it at the moment.

peterson79, 2017-10-07:
Gotcha, which compiler / IDF should I be using? This one from the readme fails to compile:

https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-59.tar.gz

I don't have the exact error at the moment... I can reproduce it if needed.

jmarcelino, 2017-10-07:
This one should work https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz

peterson79, 2017-10-07:
Yes, the compiler posted by @jmarcelino works. I still need to test a flash. I am working remotely at them moment and don't have access to the board... I will let you all know shortly.

peterson79, 2017-10-07:
Compiles, flashes and runs with:

https://dl.espressif.com/dl/xtensa-esp32-elf-osx-1.22.0-61-gab8375a-5.2.0.tar.gz

Thanks!

closed 2017-10-07 by peterson79