pycom/pycom-micropython-sigfox #79: PyCom red light (bricked) after trying to use I2C, no recovery working

HenkUyttenhove, 2017-10-08:
Please include the following information when submitting a bug report:

* The Pycom board you are using: WiPy2
* The firmware version you are using. You can get it by executing the following Python code at the 
After unboxing, did an update of firmware so should be on latest

* Exact steps to cause this issue
from machine import I2C
i2c = I2C(0, I2C.MASTER)
i2c.scan()
> result [32] and this is correct for a MCP23017 on hex 0x20
i2c.writeto_mem(32,0,0)
> Lost my serial connection
> Reboot device, no serial connection anymore (see lower)
> Tried reflash but process fails
> Tried bootsafe but red Led is active the whole time

* Outcome message after reboot:
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9010,len:12
ho 0 tail 12 room 4
load:0x3fff9020,len:388
load:0x40078000,len:11584
load:0x4009fb00,len:848
entry 0x4009fc9c

Thank you!

jmarcelino, 2017-10-08:
Do you have the MCP23017 still connected to the WiPy? Is so try disconnecting it or anything else you have connected and try again - simply communicating with it or doing the firmware upgrade again.

HenkUyttenhove, 2017-10-08:
I have deconnected everything only connecting to USB port.

I was able however with the bridge GND-G23 to get into following situation (after pushing reboot button):

Connecting on /dev/ttyUSB0...
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x3 (DOWNLOAD_BOOT(UART0/UART1/SDIO_REI_REO_V2))
waiting for download

So it looks like I can upload the firmware but don't have a copy.  Python update script not working.

jmarcelino, 2017-10-08:
You can download the current firmware from https://software.pycom.io/downloads/WiPy-1.9.0.b1.tar.gz

HenkUyttenhove, 2017-10-08:
I was able to update the firmware now (after different reboots) but still no luck.

I have the message on my screen: "your device was successfully updated to version 1.9.0.b1.  Please remove the wire and reset the board".

But still the red led after reboot.

jmarcelino, 2017-10-08:
Sounds like you Flash could be corrupted in some way. Did you ever try running a custom firmware?
Also are you running the latest firmware update tool from https://www.pycom.io/downloads/ ? There was a new one released not so long ago.

If you do have the latest version try resetting your WiPy Flash by installing esptool (instructions at https://github.com/espressif/esptool ) , doing an _erase_flash_  command with it and then running the upgrade procedure once more.

HenkUyttenhove, 2017-10-09:
@jmarcelino 
Can you please advice on the exact line I need to write the firmware.  I have found following string:

root@henk-MS-7621:/home/henk/Downloads/pyupgrade/bin#python3 esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 write_flash --flash_size 4MB 0x1000 ../../WiPy/bootloader.bin 0x10000 ../../WiPy/wipy.bin 0x4000 ../../WiPy/partitions.bin

esptool.py v2.0-beta1
Connecting...........................................

A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
root@henk-MS-7621:/home/henk/Downloads/pyupgrade/bin#

HenkUyttenhove, 2017-10-09:
Update:
After a successful flash with above command, I get following error:
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:QIO, clock div:2
load:0x3fff9010,len:12
ho 0 tail 12 room 4
load:0x04720404,len:291930117

robert-hh, 2017-10-09:
Die you try to erase flash before uploading the firmware again. Erasing is done with:

esptool.py --chip esp32 --port /dev/ttyUSB0 --baud 115200 erase_flash

And of course, you have to connect G23 (also called P2 or GPIO0) to GND & reset the device before any action with esptool or the updater.

pietrushnic, 2017-10-31:
I faced exactly same issues. My board was stuck at:

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9010,len:12
ho 0 tail 12 room 4
load:0x3fff9020,len:388
load:0x40078000,len:11584
load:0x4009fb00,len:848
entry 0x4009fc9c
```

After erasing flash with `esptool.py` and updating again wipy started to work again.

closed 2017-11-14 by HenkUyttenhove