pycom/pycom-micropython-sigfox #81: *Py: OTA firmware update incosistency

robert-hh, 2017-10-10:
There is a little hiccup involved when you mix OTA and USB flashing.
OTA uses two memory areas for the firmware and toggles these. Let's call these A and B. On bootup, the bootloader checks a flag which area is active and starts this one. The USB flash always puts the image in area A and does NOT set the "active" flag accordingly. So, if the active area is B, and you do a firmware update by USB, the new firmware will NOT be excecuted. At the moment you can do two things:
a) erase flash, which will also clear the file system
b) do another OTA, which will set again A as the acitve area. Obviously you can use the new image.

There is another unrelated hiccup when alternating between firmware images downloaded from Pycom.io and self-built images. The partition maps seem incompatible, leading to either "Red LED frozen devices" or a corrrupted file system. In that case one has to erase flash and reload with the intende method.

danicampora, 2017-10-10:
Thanks @robert-hh. The solution to both issues is to reset the OTA config area in flash when the binaries are programmed via the makefile. I will prepare that.

nevercast, 2018-01-19:
I flash with the esptool directly sometimes, otherwise I flash the firmware using FTP OTA. Esptool approach seems to work, the firmware is updated and my new code runs. the FTP OTA does not seem to work. 

Now without me derailing this thread, can you confirm if this issue I'm experience would be related to the OTA config area? If not, I'll make an issue elsewhere. 

Kind regards,
Josh.

robert-hh, 2018-01-19:
@nevercast I did not check whether and how this issue is still pertinent. When I wrote that, it seemed to be the opposite. FTP OTA worked, but depending on which image was set active, esptool.py loads would not get active. But I can re-check that. Take care, that you use appimg.bin for the OTA load.

nevercast, 2018-01-19:
The image I'm pushing, is the product of 
`make all -j4 BOARD="$BOARD" TARGET="app" FROZEN_DIR="$FROZEN_DIR" FROZEN_MPY_DIR="$FROZEN_MPY_DIR"`

Which gives me a wipy.bin. I then push wipy.bin to /flash/sys/appimg.bin

Is this correct?

robert-hh, 2018-01-19:
No. make should also create appimg.bin in the same place as wipy.bin. appimg. differs from wipy.bin in that it is extended by a signature, which is checked by the OTA loader. If that is missing or not matching, OTA will fail. If you did not change the default places for all directories, a simple
make
should be sufficient. -j4 just speeds up the make process.

nevercast, 2018-01-21:
Thanks @robert-hh, I'll copy the appimg out instead. Is the signature at the end of the application binary, i.e. Can I still use esptool also?

robert-hh, 2018-01-21:
Why would you use esptool.py for OTA?

nevercast, 2018-01-21:
I would not use esptool for OTA, I'd use it for serial flashing. 

Do I need to still use wipy.bin for esptool, or will appimg.bin work as expected?

nevercast, 2018-01-21:
Just to clarify, I'm now copying appimg.bin instead of wipy.bin. Using esptool to copy appimg works as expected, the firmware is modified. However copying appimg to /flash/sys/appimg.bin does not successfully modify the firmware. The firmware previously flashed with esptool is executed instead.

nevercast, 2018-01-29:
Update to this, I wrote a module to give me insight in to the updater and the boot info, this allowed me to notice that sometimes the update was considered finished before the ftp push had completed. Wireshark shows TCP Ack Dupes and packet drops. At this point, the issue might lay outside the updater code. I'll clarify that if I get a more certain cause.

robert-hh, 2018-01-29:
The initial issue is not related to transmission errors. It also happen if the image is copied to the SD card beforehand. And the image loaded by OTA is always activated. It's just the image loaded by USB which might not run, if the "active image" pointer points at the second partition.

nevercast, 2018-01-29:
Okay. Seems I agree with you, my issue is not related to this one. I'll raise another issue when I know more. Cheers

nevercast, 2018-05-02:
I'm no longer seeing an appimg.bin, presumably, application.mk has changed and this file is no longer created. Do I still need it over the old wipy.bin?

robert-hh, 2018-05-02:
May I close the issue then?

nevercast, 2018-05-02:
I'd say go ahead and close this, I'm about to try the new OTA image in a moment if that fails I'll just open a new issue. Thanks for your help team.

nevercast, 2018-05-02:
Looks like flashing is working as expected, couple changes such as _main and _boot, but that's easily worked around. Thanks again.

closed 2018-05-03 by catalinio

robert-hh, 2018-08-12:
I just gave it a try, and it looks as if the hiccup is still present.
a) loaded an image using OTA -> Image boots
b) loaded a new image with Makefile (make flash) ->Still the OTA image boots.
c) using safeboot -> the image loaded in step b) boots.