pycom/pycom-micropython-sigfox #83: Error building image for esp32

jingai, 2017-10-11:
Everything compiles fine until it goes to build the image:

```
% cd esp32
% make BOARD=LOPY LORA_BAND=USE_BAND_915 TARGET=app
[snipped]
Building partitions from lib/partitions.csv...
IMAGE build/LOPY_915/release/lopy_915.bin
esptool.py v2.0.1

A fatal error occurred: Segment loaded at 0x401b274c lands in same 64KB flash mapping as segment loaded at 0x401b2734. Can't generate binary. Suggest changing linker script or ELF to merge sections.
application.mk:371: recipe for target 'build/LOPY_915/release/lopy_915.bin' failed
```

robert-hh, 2017-10-11:
This is recently a common problem with the extensa gcc compiler. You have to use an older version of the compiler. See this discussion with links at the end: https://forum.pycom.io/topic/1429/solved-having-issues-build-firmware-from-soure

virtualguy, 2017-10-11:
> build-firmware-from-soure

Using the correct compiler for a given firmware seems to be a recurring
issue. Would adding a check / warning to the makefile be worthwhile? There
are already similar checks for the esp-idf and would ensure consistent
results for users  which is especially important as the compiler is
changing a lot.

jingai, 2017-10-11:
@robert-hh thanks, but that doesn't seem to solve my particular issue.  I've installed the version of the compiler referenced in that post and still get:

```
IMAGE build/LOPY_915/release/lopy_915.bin
esptool.py v2.0.1

A fatal error occurred: Segment loaded at 0x401b274c lands in same 64KB flash mapping as segment loaded at 0x401b2734. Can't generate binary. Suggest changing linker script or ELF to merge sections.
application.mk:371: recipe for target 'build/LOPY_915/release/lopy_915.bin' failed
```

robert-hh, 2017-10-11:
Which version of the C-Compiler are you actually using?

jingai, 2017-10-11:
I managed to get it to build with 1.22.0-61-gab8375a.  Thanks again.

Xykon, 2017-10-11:
I have pushed a new version of README.md to reflect the correct compiler version... closing this issue.

closed 2017-10-11 by Xykon