pycom/pycom-micropython-sigfox #84: Cant set dr_max>3 in add_channel method

Kiogora, 2017-10-11:
(sysname='LoPy', nodename='LoPy', release='1.9.0.b1', version='v1.8.6-776-gd01a46cd on 2017-09-29', machine='LoPy with ESP32', lorawan='1.0.0')

I am trying to set the network as per our multitech gateway settings in global_config.json by allowing adaptive data rates which change data rates from DR0-DR5. Unfortunately this cant be set though max dr from the documentation is 7 of which I believe DR7 is not implemented.

Kindly help.

Regards,
Alois.

danicampora, 2017-10-11:
Hi, we only support up to DR6 at the moment. What is your problem exactly?

danicampora, 2017-10-11:
Can you copy/paste the exact way that you are calling the add_channel method?

Kiogora, 2017-10-11:
I am trying to set the DR_min at 0 and DR_max at 5 because that is what the the things network json config file supports. 
Here is a snippet:
lora = LoRa(mode=LoRa.LORAWAN, public=1,  adr=1, tx_retries=0)

# Remove default channels
for index in range(0, 72):
    lora.remove_channel(index)

lora.add_channel(3, frequency=867100000, dr_min=0, dr_max=5)
lora.add_channel(4, frequency=867300000, dr_min=0, dr_max=5)
lora.add_channel(5, frequency=867500000, dr_min=0, dr_max=5)
lora.add_channel(6, frequency=867700000, dr_min=0, dr_max=5)
lora.add_channel(7, frequency=867900000, dr_min=0, dr_max=5)

#Single SF channel-DR6/SF7
#lora.add_channel(8, frequency=868300000, dr_min=6, dr_max=6)

#Join TTN Network via OTAA
lora.join(activation=LoRa.OTAA, auth=(app_eui, app_key), timeout=0)

jmarcelino, 2017-10-11:
You don't need to set the channels when using OTAA activation, the network will set them correctly them for you

Kiogora, 2017-10-11:
I have tried sensing for lora modulated signals over the air using my SDR with the automatic method you mention and I have failed to sense any. This is the only way that I can get a lora modulated signal into the air. That is why I am doing it manually. Probably this is another issue on its own?

Kiogora, 2017-10-11:
This doesn't work: https://docs.pycom.io/chapter/tutorials/lopy/lorawan-otaa.html

The firmware version above is from the pycom download page.

jmarcelino, 2017-10-11:
Is your LoPy region set correctly (via the update tool) to a EU868 band country? Where are you based?

Kiogora, 2017-10-11:
I am based in Kenya so the EU_868 applies for us. You mean the update tool from the download page? I used the linux version and I saw no such option if I recall but let me recheck.

danicampora, 2017-10-11:
@Kiogora looks like you programmed your LoPy with the wrong band (915) instead of 868.

danicampora, 2017-10-11:
@Kiogora you probably clicked next all the time without looking and the first country on the list (Argentina) was used. Let me know your WiFi MAC address and we will remove your LoPy from our servers so that you can select the country again. Thanks.

Kiogora, 2017-10-11:
Hi,

Thank you for the solution. That could be the reason as somebody had it prior.

I will revert with the mac in afew.

Kiogora, 2017-10-11:
Hi,

Which mac address do you use for registration? Does one obtain it using esptool read_mac? Or is there a micro-pythonic way to do this?

danicampora, 2017-10-11:
```python
from network import WLAN
import binascii

print(binascii.hexlify(WLAN().mac()))
```

Kiogora, 2017-10-11:
240ac400092a

Kiogora, 2017-10-11:
@danicampora Thank you. Just let me know when I can update again.

Kiogora, 2017-10-11:
240ac400092a

Thank you. Just let me know when I can update and test. I will revert with the results.

Kiogora, 2017-10-11:
240ac400092a

Thank you. Just let me know when I has been cleared from the server. I will revert with the results.

danicampora, 2017-10-11:
@Kiogora done, you can try the update again. Please let us know.

Kiogora, 2017-10-11:
Thank you. Its ok now.

closed 2017-10-11 by Kiogora