pycom/pycom-micropython-sigfox #88: L01 instability

dmartauz, 2017-10-19:
I encountered some instability of L01 while powered by my custom board. Errors are thrown when WiFi client connects to the WiFi AP of L01. Then I found out that it only happens with release 1.7.6.b1 and later. I soldered another L01 to the OEM baseboard and flashed it with 1.9.2.b1. This is the output:
`ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x12 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9010,len:12
ho 0 tail 12 room 4
load:0x3fff9020,len:388
load:0x40078000,len:11584
load:0x4009fb00,len:848
entry 0x4009fc9c
MicroPython v1.8.6-796-g489fafa0 on 2017-10-15; LoPy with ESP32
Type "help()" for more information.
>>> os.uname()
(sysname='LoPy', nodename='LoPy', release='1.9.2.b2', version='v1.8.6-796-g489fafa0 on 2017-10-15', machine='LoPy with ESP32', lorawan='1.0.0')
>>> import network
>>> wlan = network.WLAN()
>>> wlan.init(mode=network.WLAN.AP, ssid='test', auth=(network.WLAN.WPA2,'testtest'), channel=10, antenna=network.WLAN.INT_ANT)
>>> server = network.Server()
>>> Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4014597f  PS      : 0x00060a30  A0      : 0x8014e2f7  A1      : 0x3ffb6c20
A2      : 0x3ffdb4c4  A3      : 0x000000a7  A4      : 0x00000001  A5      : 0x00000000
A6      : 0x3ffb6040  A7      : 0x00000001  A8      : 0x00000001  A9      : 0x000000a7
A10     : 0x3ffb6c20  A11     : 0x00000002  A12     : 0x00000001  A13     : 0x3ffde21e
A14     : 0x0000000e  A15     : 0x0000009d  SAR     : 0x00000019  EXCCAUSE: 0x0000001c
EXCVADDR: 0x000000d9  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0xffffffff

Backtrace: 0x4014597f:0x3ffb6c20 0x4014e2f4:0x3ffb6c40 0x4014e300:0x3ffb6c60 0x40143335:0x3ffb6c80

Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40089d09  PS      : 0x00060033  A0      : 0x80088b46  A1      : 0x3ffb6860
A2      : 0x3ffb68b0  A3      : 0x3ffb6880  A4      : 0x00000000  A5      : 0x3ffc482c
A6      : 0x00000000  A7      : 0x3ffc4848  A8      : 0x00000012  A9      : 0x3ffb6830
A10     : 0x3ffb68f8  A11     : 0x3ffb6880  A12     : 0x3ffb76c0  A13     : 0x3ffb6c40
A14     : 0x3ffdb4c4  A15     : 0x3ffdbc74  SAR     : 0x00000012  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x40089d09:0x3ffb6860 0x40088b43:0x3ffb6880 0x4008ae65:0x3ffb68b0 0x4008b1da:0x3ffb6a80 0x4008aae5:0x3ffb6ac0 0x4008ac60:0x3ffb6b40 0x40081672:0x3ffb6b60 0x4014597c:0x3ffb6c20 0x4014597c:0x3ffb6c40 0x4014e300:0x3ffb6c60 0x40143335:0x3ffb6c80

Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x40089d09  PS      : 0x00060033  A0      : 0x80088b46  A1      : 0x3ffb64a0
A2      : 0x3ffb64f0  A3      : 0x3ffb64c0  A4      : 0x00000000  A5      : 0x3ffc482c
A6      : 0x00000000  A7      : 0x3ffc4848  A8      : 0x00000012  A9      : 0x3ffb6470
A10     : 0x3ffb6538  A11     : 0x3ffb64c0  A12     : 0x3ffb76c0  A13     : 0x3ffb6880
A14     : 0x3ffb68b0  A15     : 0x00000020  SAR     : 0x00000017  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c2e0  LEND    : 0x4000c2f6  LCOUNT  : 0x00000000

Backtrace: 0x40089d09:0x3ffb64a0 0x40088b43:0x3ffb64c0 0x4008ae65:0x3ffb64f0 0x4008b1da:0x3ffb66c0 0x4008aae5:0x3ffb6700 0x4008ac60:0x3ffb6780 0x40081672:0x3ffb67a0 0x40089d06:0x3ffb6860 0x40089d06:0x3ffb6880 0x4008ae65:0x3ffb68b0 0x4008b1da:0x3ffb6a80 0x4008aae5:0x3ffb6ac0 0x4008ac60:0x3ffb6b40 0x40081672:0x3ffb6b60 0x4014597c:0x3ffb6c20 0x4014597c:0x3ffb6c40 0x4014e300:0x3ffb6c60 0x40143335:0x3ffb6c80`
Many more errors follow....

Sometimes it happens immediately after establishing WiFi connection, in another cases after some time or when some communication is performed (e.g. FTP connection).

jmarcelino, 2017-11-08:
Hi sorry for the wait, can you try the new firmware just released today please?

https://forum.pycom.io/topic/2097/new-firmware-release-1-10-0-b1-external-4mbyte-ram-support

Thank you

dmartauz, 2017-11-08:
Hi, I already tried and it throws Guru Meditation Errors immediately after boot. Daniel suspects that it is caused by flash SPI frequency and is working on a patch for the updater.

nevercast, 2018-09-14:
Did Daniel say anything more on this? We get invalid instruction occasionally (GME) that we suspect is related to SPI failing to read. It can also corrupt main.py if it fails to read it and then writes over it. We have also seen completely broken FAT with multiple entries for same file name. Know of which SPI frequency may function better?

amotl, 2019-11-13:
Dear @dmartauz and @nevercast,

if you feel lucky, you might want to try our custom build just released on [1]. More background about this is available through [2].

If we are lucky together, this will improve the stability significantly. If you will be still receiving the core dumps, I will be happy if you would share its content with us. Please just don't paste it into the comment itself but put it into a file which you will be able to upload by dragging it into the text area.

Please be aware that when upgrading to the current 1.20.1 development release these builds are based upon, you will have to erase your device completely before flashing in order to keep things straight. You will find respective references to this on the forum. Hint: Use `pycom-fwtool-cli --port /dev/ttyUSB0 erase_all`, see also [3].

Please also be aware that this procedure will also erase the LoRa MAC stored on the device. @robert-hh thankfully outlined the procedure to restore it appropriately:

> - If you still know the value the LoraMac had before, then you can follow the procedure explained at https://forum.pycom.io/topic/1272/lora-mac-ffffffffffffffff/21.
> - If you do not know it, you will have to use the Pycom updater to restore it:
>   - First, use the Pycom updater (GUI version) to install a recent image (like pybytes 1.20.1) from the Internet. You do not have to use it, the installation process itself already restores both the LoRa MAC and the Sigfox ID.
>   - Then, use the Pycom updater to flash the custom image. Do not erase the flash in between.

As it turned out to gain more robustness for others already [4,5], we will be happy to learn if this happens to you as well.

With kind regards,
Andreas.

[1a] https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/LoPy-1.20.1.r1-0.6.0-vanilla-dragonfly.tar.gz
[1b] https://packages.hiveeyes.org/hiveeyes/foss/pycom/vanilla/WiPy-1.20.1.r1-0.6.0-vanilla-dragonfly.tar.gz
[2] https://community.hiveeyes.org/t/investigating-random-core-panics-on-pycom-esp32-devices/2480
[3] https://community.hiveeyes.org/t/installing-the-recent-pycom-firmware-1-20-1-r1-requires-erasing-the-flash-memory-completely/2688
[4] https://community.hiveeyes.org/t/testing-the-custom-dragonfly-builds-on-pycom-devices/2746
[5] https://github.com/pycom/pycom-micropython-sigfox/issues/361#issuecomment-553399627