pycom/pycom-micropython-sigfox #90: Can't find builtins.compile() function

gbeardall, 2017-10-28:
Using LoPy:
 ```
(sysname='LoPy', nodename='LoPy', release='1.9.2.b2', version='v1.8.6-796-g489fa
fa0 on 2017-10-15', machine='LoPy with ESP32', lorawan='1.0.0')
```
If i do:
```
import builtins
help(builtins)
```
it doesn't report the builtins compile() function, 
whereas: [https://docs.pycom.io/chapter/firmwareapi/micropython/builtin.html](url) suggests that there is one.

- Is this function included in this build?
- If not, was it ever or will it be in future?

Thanks

robert-hh, 2017-10-28:
It looks like it does not exist. However, you can feed a source string directly into builtins.exec().

robert-hh, 2017-10-28:
If you need it, you may make your custom firmware image and add the line
`#define MICROPY_PY_BUILTINS_COMPILE                 (1)`
to the file esp32/mpconfigport.h

gbeardall, 2017-10-29:
Ok, thanks for the suggestions