pycom/pycom-micropython-sigfox #91: Deepsleep go_to_sleep function does not work with UART(1)

affoltep, 2017-10-28:
Please include the following information when submitting a bug report:

* The Pycom board you are using (e.g. WiPy 2.0, LoPy).
**Lopy**

* The firmware version you are using. You can get it by executing the following Python code at the REPL:
```
(sysname='LoPy', nodename='LoPy', release='1.9.2.b2', version='v1.8.6-796-g489fafa0 on 2017-10-15', machine='LoPy with ESP32', lorawan='1.0.0' )
```

The following code prevents the LoPy for changing in deepsleep, issued by the initialization of the UART on port 1:

```
from machine import UART
from deepsleep import DeepSleep

ds = DeepSleep()

uart = UART(1, 9600)                         # init with given baudrate
uart.init(9600, bits=8, parity=None, stop=1) # init with given parameters
#uart.deinit()
#
ds.go_to_sleep(60)  # go to sleep for 60 seconds
```

If you uncomment the the uart.deinit() the LoPy crash.

Regards 
Peter

jmarcelino, 2017-10-28:
That's because the DeepSleep class also uses UART 1 to communicate with the DeepSleep shield
https://github.com/pycom/pycom-libraries/blob/master/deepsleep/deepsleep.py#L32

affoltep, 2017-10-30:
Many Thank jmarcelino. You are completely right. I should have study the source code before writing an issue ;)

closed 2017-10-30 by affoltep