pycom/pycom-micropython-sigfox #92: Release 1.10.0.b1: Link step fails - missing modules

robert-hh, 2017-11-08:
Building fails at load time (and yes: new compiler version, new esp-idf version) with the error:
```
LINK build/LOPY_868/release/application.elf
/home/robert/Downloads/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lpthread
/home/robert/Downloads/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/5.2.0/../../../../xtensa-esp32-elf/bin/ld: cannot find -lesp_adc_cal
collect2: error: ld returned 1 exit status
application.mk:375: recipe for target 'build/LOPY_868/release/application.elf' failed
make: *** [build/LOPY_868/release/application.elf] Error 1
```
Where's the missing stuff?

jmarcelino, 2017-11-08:
Hi @robert-hh 
Thanks for reporting this, it seems the files were not added to latest commit but I’m sure @danicampora will add them soon.

As a temporary workaround you can usually compile a suitable pycom-esp-idf example, e.g. gatt_server and copy the missing libs from the *build* folder.

Thanks again

robert-hh, 2017-11-08:
Daniel updated the repository. Thanks a lot.

closed 2017-11-08 by robert-hh