pycom/pycom-micropython-sigfox #95: [firmware] LoPy firmware flashing fails

quentin-ol, 2017-11-14:
I've received a new LoPy yesterday and attempted a firmware upgrade via the PyCom Firmware Upgrade tool. 

According to the application, the firmware upgraded happened smoothly but the version that was displayed on the `REPL` didn't match that of the firmware that was advertised by the tool (1.10.0.b1). For the record, here is what it says:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9028,len:8
load:0x3fff9030,len:932
load:0x4009fa00,len:0
ho 12 tail 0 room 4
load:0x4009fa00,len:13232
entry 0x400a04e8
MicroPython v1.8.6-820-g56004e13 on 2017-11-08; LoPy with ESP32
Type "help()" for more information.
```

Then, I've attempted using the method described [here](https://github.com/pycom/pycom-micropython-sigfox) to build the firmware from source and to flash it. Building was fine but flashing ended up with a bricked LopY (red LED). 

I managed to get the firmware working using the method described in issue #81 (i.e. first erase the flash then proceed as explain in [here](https://github.com/pycom/pycom-micropython-sigfox)). In case anyone needs it, here is the procedure:
```bash
python esptool.py --chip esp32 --port <LoPY PORT> --baud 115200 erase_flash
```

danicampora, 2017-11-19:
In order to get the actual version, please do:

```python
import os

os.uname()
```

... and check the **release** attribute.

nevercast, 2017-11-30:
Which build are you using? I'm building from eca3b714 and it seems to be bricking my WiPy 2 with the same RTC issue.

I'm not sure if anything has changed in the last 5 months as far as build process goes, but my build script worked with 5 months ago. Hopefully we are experiencing the same issue and can narrow it down.

nevercast, 2018-07-20:
@quentin-ol Have you seen this issue recently? The firmware has changed a lot and the updater has had a few changes too.

sdaniel55, 2018-09-03:
Hi 👋 
I'll close this issue because it has not had any recent activity. Thank you for your contributions!

closed 2018-09-03 by sdaniel55