pycom/pycom-micropython-sigfox #96: [LoRaWAN] Error in LoRaMacChannelManualAdd

quentin-ol, 2017-11-16:
Hello,

I've spotted an error in `LoRaMac.c`:

```c
LoRaMacStatus_t LoRaMacChannelManualAdd( uint8_t id, ChannelParams_t params )
{
  [...]
        if( ValueInRange( params.DrRange.Fields.Max, DR_5, LORAMAC_TX_MAX_DATARATE ) == false )
        {
            datarateInvalid = true;
        }
    }
[....]
```
`DR_5` should be replaced by `DR_0`. 

In the current state, this prevents forcing the LoPy to lower its max data rate.

danicampora, 2017-11-19:
Thanks, we will fix it!

dmartauz, 2017-11-29:
Does it mean that OTAA join is performed at DR_5 as well?

jmarcelino, 2017-11-29:
@dmartauz this was only related to adding channels manually.

The corrections were integrated already in https://github.com/pycom/pycom-micropython-sigfox/commit/eca3b714a11c4b84e740a36b4096a20aac3c9be2

closed 2017-11-29 by jmarcelino

dmartauz, 2017-11-29:
@jmarcelino 
Is there any documentation which data rates are used during OTAA join and what is the sequence?