pycom/pycom-micropython-sigfox #98: How to get the actual date?

liowmei, 2017-11-16:
Hi there, i am new to pycom devices and i try out the example for pytrack from the github and i cannot get the actual date but only stop at year 1970. May i know how can i solve this? Thanks in advance. 

* The Pycom board :  LoPy and pytrack 
* The firmware version : v1.9.2.b2

[image]

[image]

rolandvs, 2017-11-22:
@liowmei - this is more a question for the forum.pycom.io

As far as concerned with the "demo" code from pycom: It doesn't work from what you expect. There needs to be a network connection in order to use NTP. In the demo there is no connection made to a IP network whatsoever. So the difference between working/not working is maybe the use of a connection via WiFi to update the device instead of using an USB connection.

You could extract the date/time from the GPS info strings and use that to set the time.

IMHO - the demo shows the use of a gps tracker. As most trackers are used outside - without a wireless network like WiFi - setting the time like this is not what you want. The demo on itself needs to be uncommented at various places to store the GPS readings with your cable to USB disconnected.

rolandvs, 2017-11-22:
@liowmei - you can look in my fork @rolandvs of the Pycom-libraries for a slightly more usable demo.

liowmei, 2017-11-22:
@rolandvs  Hi, thanks for you guide. I facing problem on connecting on the lopy serial board. Before i connect the lopy+expansion board, i have connected it with pytrack+lopy which is on COM 6. After that i try to connect lopy+expansion but it show not connected and keep appearing "connecting COM6". I have also change the COM number at the setting since the lopy+expansion board is using COM7. May i know how can i fix it?

rolandvs, 2017-11-22:
@liowmei - Best to move this to `forum.pycom.io` as the issues reported here must address "real problems" directly related to the hardware and software of the pycom modules.

I think you can get more and faster answer through the forum, but the first thing that comes to my mind is that you plug it into a different USB port?

rolandvs, 2017-11-23:
@liowmei - If you take a look at my fork `rolandvs/pycom-libraries/tree/master/pytrack` of the Pycom-libraries again you will see a fixed sample for the `Pytrack` that uses `Wi-Fi` to set the `RTC` through the use of the `NTP` server. I extended the code a little so it even might be usable. :-). Have fun.

sdaniel55, 2018-09-03:
I think this has been resolved, so I'll close this issue. If you think otherwise, feel free to reopen it and add more details!

closed 2018-09-03 by sdaniel55