pycom/pycom-micropython-sigfox #99: modwlan.c: Setting ip with ifconfig() after connect() leads to core dump and device locks

robert-hh, 2017-11-17:
This simple scripts works for a freshly set-up device only once. If then a soft-reset is done with Ctrl-D, it does not connect any more, and also may cause a core dump. Erasing and reloading flash helps to reestablish WiFi. When the device is powered off, this lock state is released after a while too, indicating some timeouts which cure the lock. The issue occurred with version 1.10.1.b1. Previous versions worked fine.
```
import network
import time
# setup as a station
wlan = network.WLAN(mode=network.WLAN.STA)
wlan.connect(ssid="my_AP", auth=(network.WLAN.WPA2, "my_key"))
for _ in range (200):
    if wlan.isconnected():
        wlan.ifconfig(config=("10.0.0.141", "255.255.255.0", "10.0.0.240",  "10.0.0.240"))
        print(wlan.ifconfig())
        break
    else:
        time.sleep_ms(50)
else:
    print("Could not connect")
```
Placing the ifconfig() call before connect() or adding a delay of a second between the connect() and ifconfig call aviods the issue.
Trying to understand what's going on by looking into modwlan.c makes me wonder, what actually happens. There seems to be a condition mix-up between the local state of the wlan_obj.disconnected variable and the connection status in the wifi_event_group variable. ifconfig(), when called, disconnects first, which causes a SYSTEM_EVENT_STA_DISCONNECTED event, which attempts to reconnect, while the dhcp client is shut down and the new ip values are set, and the timing relations are not obvious. This might lead to trouble seen.
Crash log:
```
PYB: soft reboot
Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x4010df00  PS      : 0x00060e30  A0      : 0x80119e6e  A1      : 0x3ffbd590  
A2      : 0x3ffc8ca8  A3      : 0x3ffd5cb9  A4      : 0x3ffd5ce3  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x8010df00  A9      : 0x3ffbd580  
A10     : 0x3ffc8cf7  A11     : 0x00000000  A12     : 0x0000000f  A13     : 0x3ffc8d06  
A14     : 0x02ac0f00  A15     : 0x00000000  SAR     : 0x00000010  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x000000f1  LBEG    : 0x400976c4  LEND    : 0x400976cf  LCOUNT  : 0xffffffff  

Backtrace: 0x4010df00:0x3ffbd590 0x40119e6b:0x3ffbd5c0 0x401169e1:0x3ffbd610 0x4011758a:0x3ffbd690 0x40117e35:0x3ffbd760 0x401213b9:0x3ffbd780 0x40122f6f:0x3ffbd7d0

================= CORE DUMP START =================
5CcAAAoAAABsAQAA
gNj7P9DU+z942Ps/
0Nb7PxDY+z9AahtA4ED8P+BA/D+A2Ps/2ED8PwIAAAAQwvs/EML7P4DY+z8AAAAA
FwAAAHzI+z93aWZpAF4oIkY0Td71dAYAAAAAAHjY+z8AAAAAIAIGABcAAAABAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAo+vw/kPr8P/j6/D8AAAAAAAAAAAEAAAAAAAAA
h1BAPwAAAADAfAlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABTqTQ==
MCf7PwDfEEAwDgYAbp4RgJDV+z+ojPw/uVz9P+Nc/T8AAAAAAAAAAAAAAAAA3xCA
gNX7P/eM/D8AAAAADwAAAAaN/D8AD6wCAAAAABAAAAAcAAAA8QAAAMR2CUDPdglA
/////w8AAAAGjfw/ECsIQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAARV/T+w1fs/AQAAAAAAAADkaRGAwNX7P6iM/D+g1vs/
uVz9PwAAAABAAAAAGAAAAACNEYDQ1fs/qIz8P6CW/D+NdRGAENb7P6iO/D8DAAAA
zEdCPxDW+z+ojPw/K438PzIAAABMAAAAANb7PwIAAADihRGAAgAAAIJf/D84m/w/
OA66FgAAAADolvw/qI78Pzh+EYCQ1vs/KFj9P3RV/T8zMzMDzEdCPwEAAAAAAAAA
QjgyAFDW+z+PCQAAAAAAAMxHQj8oWP0/iJv8PwAAAAB2hhGAkNb7P6Bf/D9cGPw/
GQAAAEhg/D+gXP0/AQAAAAEAAAD8YPw/AwAAABkAAACojPw/OJv8P5Bc/T8IAAAA
vBMSgGDX+z90Vf0/GQAAAAAAAAAAAEAA/f///wAAAACaXP0/CAgxBAABZAAAAAAA
tFz9P4wBAAAAAAAAqFz9PwAAAAC0XP0/uVz9P+Nc/T8AAAAAZF39P8lc/T+AXf0/
6Vz9PyFd/T8AAAAApV39PwAAAAAAAAAAJArEBuOYCIBAXv0/QBIAAP////8AQfw/
EQAAABkAAAB0Vf0/AAAAAKXdEYBw1/s/4HQJQBAAAADwWf0/AAAAAAEAAAAAAAAA
AQAAAFoDAABQAAAAAQAAAHIvEoCA1/s/sAMAALSh/D8AAAAAkFz9P5Bc/T8AAWAB
AAAAANDX+z+wofw/CGP8PwAAAABYb/w/sKH8Pwhj/D8AAAAAAAAAALCh/D+8DLoW
/////9DX+z+wofw/CGP8P3Rc/T8kfhFA9QAAAAQAAAAAAAAAENj7PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAABEAAAAAAAAAAAAAAAAAAAAaAAAAby8SQAS4QT8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAHNj7PwAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
```

robert-hh, 2017-11-18:
if juggled a little bit with this phenomenon, and it could be caused by the esp_wifi_disconnect() in if_config() at line 956 in mods/modnetwork.c.
```
            if (adapter_if == TCPIP_ADAPTER_IF_STA) {
                // first close any active connections
                esp_wifi_disconnect();
                tcpip_adapter_dhcpc_stop(TCPIP_ADAPTER_IF_STA);
                tcpip_adapter_set_ip_info(adapter_if, &ip_info);
                tcpip_adapter_set_dns_info(adapter_if, TCPIP_ADAPTER_DNS_MAIN, &dns_info);
            } else {
                tcpip_adapter_dhcps_stop(TCPIP_ADAPTER_IF_AP);
                tcpip_adapter_set_ip_info(adapter_if, &ip_info);
                tcpip_adapter_set_dns_info(adapter_if, TCPIP_ADAPTER_DNS_MAIN, &dns_info);
                tcpip_adapter_dhcps_start(TCPIP_ADAPTER_IF_AP);
            }
```
If I comment that disconnect staement out, everything works well, and the IP is still changed. 
I added a print statement to the wlan_event_handler() to see the sequence of events. That shows that even if this disconnect statement is left off, the connection is disconnected and re-connected when required.
Besides that, setting the IP address first and connecting then looks in any case as a robust approach.

danicampora, 2017-11-24:
Should be fixed with the latest release.

robert-hh, 2017-11-24:
ACK

closed 2017-11-24 by robert-hh