pycom/pycom-micropython-sigfox #108: Pycom frozen MQTTClient leaks memory when publishing messages under 1.11.0.b1

rkeiii, 2017-12-20:
I'm seeing this on a WiPy 2.0 running my own build of 1.11.0.b1. The only changes I made to the firmware was to add datetime into the frozen modules area. In the test case below I didn't even load datetime so I believe that's irrelevant. This same memory leak seems to occur in the last release of the firmware as well when the frozen MQTTClient was introduced (at least that's how it looked to me).

```
(sysname='WiPy', nodename='WiPy', release='1.11.0.b1', version='063e199 on 2017-12-19', machine='WiPy with ESP32')
```
* Exact steps to cause this issue
    1. Modify the boot.py and main.py test case below with your own MQTT broker URL, user, pass, WiFi creds, etc.
    2. Transfer the main.py and boot.py onto a WiPy 2.0 running 1.11.0.b1
    3. Reset the WiPi and let it loop sending messages until it runs out of RAM

* Here's the main.py
```
import MQTTConst
import micropython
import gc

from MQTTClient import MQTTClient
from machine import Timer

class HeartbeatTimer:
    def __init__(self):
        self._mqtt_client = MQTTClient("tester", True, MQTTConst.MQTTv3_1_1)

        self._mqtt_host = '192.168.1.66'
        self._mqtt_port = 1883
        self._mqtt_topic = 'realtime'
        self._mqtt_user = 'myusername'
        self._mqtt_pass = 'secr3t'
        self._mqtt_qos = 0

        self._iters = 0

        self._mqtt_client.configEndpoint(self._mqtt_host, self._mqtt_port)
        self._mqtt_client._user = self._mqtt_user
        self._mqtt_client._password = self._mqtt_pass
        self._mqtt_client.connect()

        self._alarm = Timer.Alarm(self._wakeup_handler, arg=self, s=0.25, periodic=True)

    def _wakeup_handler(self, arg):
        self._iters += 1
        print("iteration", self._iters)
        self._mqtt_client.publish("events", "hello!!", 0, False)
        micropython.mem_info()
        gc.collect()

print("Starting up...")

# software versions
print("Version Information: ", os.uname())

# create our timer
hbtimer = HeartbeatTimer()

# startup is complete so let's take a nap and wait for something to wake us
machine.sleep()
```

And here's the boot.py
```
# boot.py -- run on boot-up
import os
import machine
from machine import UART
from network import WLAN

# ???
uart = UART(0, 115200)
os.dupterm(uart)

[... 177 lines cut ...]
 No. of 1-blocks: 914, 2-blocks: 818, max blk sz: 512, max free sz: 6
iteration 856
Unhandled exception in callback handler
Traceback (most recent call last):
  File "main.py", line 32, in _wakeup_handler
  File "MQTTClient.py", line 161, in publish
MemoryError: 
iteration 857
Unhandled exception in callback handler
Traceback (most recent call last):
  File "main.py", line 32, in _wakeup_handler
  File "MQTTClient.py", line 161, in publish
MemoryError: 
Unhandled exception in thread started by <bound_method>
Traceback (most recent call last):
  File "MQTTMsgHandler.py", line 257, in _io_thread_func
  File "MQTTMsgHandler.py", line 232, in _verify_connection_state
  File "MQTTMsgHandler.py", line 210, in _send_pingreq
  File "MQTTMsgHandler.py", line 150, in priority_send
  File "MQTTMsgHandler.py", line 223, in _send_packet
  File "MQTTMsgHandler.py", line 220, in _send_packet
MemoryError: 
iteration 858

```

rkeiii, 2017-12-20:
Here's an even shorter main.py removing the class entirely.

```
import MQTTConst
import micropython
import gc

from MQTTClient import MQTTClient
from machine import Timer

mqtt_client = MQTTClient("tester", True, MQTTConst.MQTTv3_1_1)

def wakeup_handler(none):
    global mqtt_client
    mqtt_client.publish("events", "hello!!", 0, False)
    micropython.mem_info()
    gc.collect()

print("Starting up...")

# software versions
print("Version Information: ", os.uname())

mqtt_host = '192.168.1.66'
mqtt_port = 1883
mqtt_topic = 'realtime'
mqtt_user = 'user'
mqtt_pass = 'pass'
mqtt_qos = 0

mqtt_client.configEndpoint(mqtt_host, mqtt_port)
mqtt_client._user = mqtt_user
mqtt_client._password = mqtt_pass
mqtt_client.connect()

alarm = Timer.Alarm(wakeup_handler, arg=None, s=0.25, periodic=True)

# startup is complete so let's take a nap and wait for something to wake us
machine.sleep()

```

rkeiii, 2017-12-20:
I replaced the Pycom provided MQTTClient with umqtt.simple from micropython-lib and the issue is gone.

Here's the test using umqtt.simple...
```
#import MQTTConst
import micropython
import gc
from mqtt import MQTTClient

#from MQTTClient import MQTTClient
from machine import Timer

mqtt_host = '192.168.1.66'
mqtt_port = 1883
mqtt_topic = 'realtime'
mqtt_user = 'auser'
mqtt_pass = 'apass'
mqtt_qos = 0

#mqtt_client = MQTTClient("tester", True, MQTTConst.MQTTv3_1_1)
mqtt_client = MQTTClient("umqtt_client", mqtt_host)

def wakeup_handler(none):
    global mqtt_client
    mqtt_client.publish(b"events", b"hello!!")
    micropython.mem_info()
    gc.collect()

print("Starting up...")

# software versions
print("Version Information: ", os.uname())

#mqtt_client.configEndpoint(mqtt_host, mqtt_port)
#mqtt_client._user = mqtt_user      
#mqtt_client._password = mqtt_pass
#mqtt_client.connect()

mqtt_client.user = mqtt_user
mqtt_client.pswd = mqtt_pass
mqtt_client.connect()

alarm = Timer.Alarm(wakeup_handler, arg=None, s=0.25, periodic=True)

# startup is complete so let's take a nap and wait for something to wake us
machine.sleep()
```

Here's the beginning of an example run...
```
rst:0x7 (TG0WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9028,len:8
load:0x3fff9030,len:1060
load:0x4009fa00,len:0
ho 12 tail 0 room 4
load:0x4009fa00,len:15160
entry 0x400a0644
Starting up...
Version Information:  (sysname='WiPy', nodename='WiPy', release='1.11.0.b1', version='063e199 on 2017-12-19', machine='WiPy with ESP32')
MicroPython 063e199 on 2017-12-19; WiPy with ESP32
Type "help()" for more information.
>>> stack: 688 out of 5120
GC: total: 67008, used: 17728, free: 49280
 No. of 1-blocks: 147, 2-blocks: 23, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
GC: total: 67008, used: 15552, free: 51456
 No. of 1-blocks: 75, 2-blocks: 14, max blk sz: 384, max free sz: 2284
stack: 688 out of 5120
```

oligauc, 2017-12-27:
This is not a leakage but your output queue is growing till no memory is left. In your logs i cannot see the line "Packet sent. (Length: XXXX)". Please check the connection to your broker. 

The library you are using is AWS compliant and you should interface with the MQTTLib.py file. Please see this link for a usage example: [https://docs.pycom.io/chapter/tutorials/all/aws.html] 

If you use the publishing example you should something like this:

Packet sent. (Length: 4)
Iteration  1996  GC free  31104
Packet sent. (Length: 34)
Received a new message:
b'New Message 1996'
from topic:
b'PublishTopic'
--------------

Iteration  1997  GC free  29504
Packet sent. (Length: 4)
Iteration  1998  GC free  29056
Packet sent. (Length: 2)
Packet sent. (Length: 34)
Packet sent. (Length: 34)
Received a new message:
b'New Message 1997'
from topic:
b'PublishTopic'
--------------

rkeiii, 2017-12-27:
Hello @oligauc,

Thank you for pointing out that I was improperly accessing MQTTClient. I'm still learning where all the documentation is.

I've conducted further testing and amended the code to address the points you raised. I believe I'm still seeing the same issue I originally raised and I also have a design concern.

Design Concern: If you connect to a non-existent host in the first place the Pycom MQTT client doesn't seem to raise any exception at all. Am I missing something here? I would also tend to expect client software to send something like ECONNRESET (or whatever) in the event the connection disappears mid-run.

Original Issue: Client still seems to leak memory for some reason even with mqtt_client.configureOfflinePublishQueueing(0) and mqtt_client.configureDrainingFrequency(10). I would expect this to disable publish queuing or at the very least try to empty the queue every 1/10th of a second (and my code only try's to send a message every quarter of a second).

Here's my amended test case code that addresses your points. I ran it against a RabbitMQ broker in my testing.
```
import MQTTConst
import micropython
import gc

from MQTTLib import AWSIoTMQTTClient as MQTTClient
from machine import Timer

mqtt_client = MQTTClient("tester")

def wakeup_handler(none):
    global mqtt_client
    mqtt_client.publish("events", "hello!!", 0)
    connection_state = mqtt_client._mqttClient._msgHandler.isConnected()
    print("mqtt client connected?: %s" % connection_state)
    micropython.mem_info()
    gc.collect()

print("Starting up...")

# software versions
print("Version Information: ", os.uname())

mqtt_host = '192.168.1.66'
mqtt_port = 1883
mqtt_topic = 'cantbelieve'
mqtt_user = 'donaldtrump'
mqtt_pass = 'ispresident'
mqtt_qos = 0

mqtt_client.configureOfflinePublishQueueing(0)
mqtt_client.configureDrainingFrequency(10)
mqtt_client.configureEndpoint(mqtt_host, mqtt_port)
conn = mqtt_client._mqttClient
conn._user = mqtt_user
conn._password = mqtt_pass

if conn.connect():
    print('MQTT connection succeeded')

alarm = Timer.Alarm(wakeup_handler, arg=None, s=0.25, periodic=True)

# startup is complete so let's take a nap and wait for something to wake us
machine.sleep()
```

Here's the beginning and end of a sample run...
```
Starting up...
Version Information:  (sysname='WiPy', nodename='WiPy', release='1.11.0.b1', version='7200c89-dirty on 2017-12-25', machine='WiPy with ES)
Packet sent. (Length: 44)
MQTT connection succeeded
MicroPython 7200c89-dirty on 2017-12-25; WiPy with ESP32
Type "help()" for more information.
>>> mqtt client connected?: True
stack: 704 out of 5120
GC: total: 67008, used: 32128, free: 34880
 No. of 1-blocks: 270, 2-blocks: 37, max blk sz: 512, max free sz: 2177
mqtt client connected?: True
[... 219 lines cut ...]
  File "main.py", line 12, in wakeup_handler
  File "MQTTLib.py", line 50, in publish
  File "MQTTClient.py", line 161, in publish
MemoryError: 
Unhandled exception in callback handler
Traceback (most recent call last):
  File "main.py", line 12, in wakeup_handler
  File "MQTTLib.py", line 50, in publish
  File "MQTTClient.py", line 161, in publish
MemoryError: 
Unhandled exception in thread started by <bound_method>
Traceback (most recent call last):
  File "MQTTMsgHandler.py", line 262, in _io_thread_func
  File "MQTTMsgHandler.py", line 223, in _send_packet
  File "MQTTMsgHandler.py", line 220, in _send_packet
MemoryError: 
Unhandled exception in callback handler
Traceback (most recent call last):
  File "main.py", line 12, in wakeup_handler
  File "MQTTLib.py", line 50, in publish
  File "MQTTClient.py", line 161, in publish
MemoryError:
```

I'm definitely seeing packets get sent in my output but it doesn't seem to occur at the frequency I would expect. Did I not disable queueing/draining properly?

oligauc, 2017-12-28:
The MQTT library requires 3 types of connections:

1. Wifi connection to your local router
2. Socket connection to the server
3. Connection to the broker

In the first 2 types, an exception is thrown if something goes wrong. In the 3rd type, conn.connect() will return false

The Pycom MQTT library was designed to work with AWS (Amazon IoT). It takes about 0.05 sec to send a publish request, process the reply and call the callback. You can test this with the chronometer. 

In your case you are using the RabbitMQ. RabbitMQ is replying to the connect and ping requests, but it is not sending a reply to the publish request. This is making the receive timeout at 3 sec. If you are connected, the queue does not drop messages. You need to configure the maximum queue size, with a non zero value, to start dropping messages.

rkeiii, 2017-12-28:
@oligauc 

I don't understand what you're trying to say about there being "3 connections". To the best of my knowledge the socket connection to the server is made directly to RabbitMQ. This _is_ the connection to the broker so there are in fact only two connections, right? In my case they are...

1. Wifi connection from my WiPy 2.0 to my local router
2. Socket connection from my WiPy 2.0 to my local RabbitMQ instance

Regarding the design concern: The client behavior seems extremely odd to me. If you feed the Pycom MQTT client an invalid MQTT host to connect to it will "pretend to connect" and just loop infinitely never even notifying the user there's a problem. This behavior really doesn't seem right to me. Why would this code not throw a sane error like "host not found" or "connection timed out"? It seems wrong for the end user to have to check conn.connect() and then throw a meaningful error because the Pycom code does not. Perhaps I should open a separate issue to address this.

Regarding the original issue: When you say "You need to configure the maximum queue size, with a non zero value, to start dropping messages." are you referring to ie mqtt_client.configureOfflinePublishQueueing(1)?

oligauc, 2017-12-28:
The 3 rd connection is a packet. When you call conn.connect() a socket is set up and a connect packet is sent to the MQTT server. 

If you try to set up a socket connection to a socket server an exception is thrown.

rkeiii, 2017-12-28:
@oligauc I see no reply at all regarding the design concern? Would you prefer I open a separate issue? Does it seem reasonable to you to expect the SocketError to not be suppressed?

This file: https://github.com/pycom/pycom-micropython-sigfox/blob/eca3b714a11c4b84e740a36b4096a20aac3c9be2/esp32/frozen/MQTTMsgHandler.py

At lines 92-99 the socket error is suppressed and not re-thrown. This hides the connection failure from upstream code unless the upstream code explicitly evaluates the return, this does not seem right to me. Why would it be desirable to return a boolean defining success/failure instead of letting the socket error bubble up?

oligauc, 2017-12-29:
If the connection to the server is lost (e.g server down) the device re-attempts to connect. Ideally you want the device to recover from any failure instead of just stop working.

rkeiii, 2017-12-29:
I definitely agree having the client try to re-establish the connection is a good idea. However, it is not necessary to _permanently_ suppress the exception in order to attempt to re-establish the connection. The code could, for example, try to re-establish the connection N times (N being configurable) and then allow the exception to bubble up.

danicampora, 2017-12-29:
> The code could, for example, try to re-establish the connection N times (N being configurable) and then allow the exception to bubble up.

I agree, it's a good idea. @oligauc can we do that? Add a maximum number of retries? Thanks.

oligauc, 2017-12-29:
@danicampora , @rkeiii 

OK i will implement it for the next release