pycom/pycom-micropython-sigfox #111: Timer.Alarm does not immediately update free timer count after alarm.callback(None)

rkeiii, 2017-12-22:
Please include the following information when submitting a bug report:

* The Pycom board you are using: WiPy 2.0
* The firmware version you are using: custom build of 1.11.0.b1
```
Version Information:  (sysname='WiPy', nodename='WiPy', release='1.11.0.b1', version='68057a7 on 2017-12-22', machine='WiPy with ESP32')
```

* Exact steps to cause this issue
    1. Create 16 Timer.Alarm objects 
    2. Call alarm.callback(None) on several of them in rapid succession
* What you expected
    * I expected to no longer receive "MemoryError: maximum number of 16 alarms already reached" after calling alarm.callback(None) on a Timer.Alarm object 
* What happened instead
    * I received "MemoryError: maximum number of 16 alarms already reached" for some number of seconds before the Timer.Alarm code finally realized there were some free timers.

Here's the "timer code" that I'm using the instantiate the Timer.Alarm's and also to free them with alarm.callback(None).
```
import config
import util
import _thread
import time
import msg
import gc

from machine import Timer
from exception import TimersSoftLimitException, TimersHardLimitException
from uasyncio.queues import Queue
from uasyncio.queues import QueueEmpty

# TODO: This looks like insanity because it is.
# The first time you load datetime it errors with
# AttributeError: type object 'tzinfo' has no attribute '__new__'
# Upon the second load it load fine!?
try:
    from datetime import datetime
except AttributeError:
    pass
from datetime import datetime

timers = []
timers_lock = _thread.allocate_lock()
task_queue = Queue(maxsize=config.queued_tasks_limit)

class BaseTimer:
    """Base timer class"""

    def __init__(self, wakeup_callback, sleep_us, is_periodic):
        try:
            with timers_lock:
                if len(timers) < config.timers_hard_limit:
                    gc.collect()
                    self._alarm = Timer.Alarm(wakeup_callback, ms=sleep_us, arg=None, periodic=is_periodic)
                    timers.append(self)
                    print("timers in use:", len(timers))
                else:
                   raise TimersHardLimitException()
        except MemoryError as e:
            print("Probably spurious MemoryError suppressed: ", e)

class EtaTaskCallTimer(BaseTimer):
    """Timer to supprt ETA type Celery Task calls"""

    def __init__(self, task):
        # calculate how to long to sleep based on Task ETA
        now = datetime(*config.rtc.now())
        tdiff = task.eta - now
        # TODO: No day handling. This is fucked just as soon as we try to time something longer than "minutes" wtf that means
        sleep_us = (tdiff.seconds * 1000) + (tdiff.microseconds / 1000)

[... 121 lines cut ...]
(2017, 12, 22, 23, 26, 42, 455621, None)  task 97d3f8d4-c6a7-4fb8-a8f3-5763bc8b25e7 completed successfully with 0:00:00.012633 deviation
(2017, 12, 22, 23, 26, 42, 552617, None)  task 0443d8a1-006b-4640-8066-2f14a1d2f392 completed successfully with 0:00:00.009632 deviation
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
(2017, 12, 22, 23, 26, 42, 655610, None)  task 0ed33b33-e4ab-420f-aa05-4b1fd845a99e completed successfully with 0:00:00.012632 deviation
(2017, 12, 22, 23, 26, 42, 756610, None)  task 653c2e0a-918e-4841-a368-93ffe83ceaff completed successfully with 0:00:00.013637 deviation
(2017, 12, 22, 23, 26, 42, 855612, None)  task 90f58dec-6c55-4793-bb71-784987894d0d completed successfully with 0:00:00.012647 deviation
(2017, 12, 22, 23, 26, 42, 955627, None)  task 0f41de6c-16dc-4d26-80fa-ef16d2c3ac52 completed successfully with 0:00:00.012674 deviation
(2017, 12, 22, 23, 26, 43, 52599, None)  task 6218b69f-3a67-4204-bf26-e4a1229540e0 completed successfully with 0:00:00.009660 deviation
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
(2017, 12, 22, 23, 26, 43, 152563, None)  task 3f881c96-4012-4fc4-b876-8b49ca15a308 completed successfully with 0:00:00.009580 deviation
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
Probably spurious MemoryError suppressed:  maximum number of 16 alarms already reached
...........................................
```

rkeiii, 2017-12-23:
I realized after doing some more testing with a friend that I needed to be calling Timer.Alarm.cancel() instead of Timer.Alarm.callback(None).

This page in the documentation seems unclear about how to properly disregard a Timer you're no longer interested in. Am I missing something?

Could I submit a documentation patch somehow?

rkeiii, 2017-12-23:
Sorry, here's the docs page I was talking about: https://docs.pycom.io/chapter/tutorials/all/timers.html

robert-hh, 2017-12-23:
@rkeiii That's right. It used to be, that  Timer.Alarm.callback(None) would remove the alarm. But now it just re-enables the alarm. The section of interest is int mods/machtimer_alarm.c, line 237. replacing line 238 by
```
alarm_delete(self_in); // that is the function serving alarm.cancel()
return;
```
could do the trick.

rkeiii, 2017-12-24:
@robert-hh I'm quite new to Micropython as a whole so I'd be open to suggestion on whichever approach would be better. My gut tells me completely cancelling the Timer.Alarm as a result of calling Timer.Alarm.callback(None) might be somewhat non-intuitive behavior. What if I want to re-use the Timer.Alarm with a new callback? So I think I would probably lean toward improving the documentation but like I say I'm no authority on the matter.

Would you happen to know if it's possible/where to submit a documentation patch?

robert-hh, 2017-12-24:
@rkeiii Since the timer has only the noticeable reaction of calling a callback, Timer.Alarm.callback(None) is practically the same as Timer.Alarm.cancel(). There is not need to let the timer running if there is nothing to do.
The place for the documentation is here: https://github.com/pycom/pycom-documentation
You could place a Pull-request there or an issue, although Pycom is not very active in dealing with pull requests. They typically take what's suggested there and copy it into their private repository. Maybe they it's because git is not a very user-friendly tool.

danicampora, 2017-12-26:
Hello,

Thanks for reporting this. ``Timer.Alarm.callback(None)`` should cancel the alarm as well. The documentation is correct, there's a bug in the code.

Cheers,
Daniel

danicampora, 2017-12-26:
The fix is actually to change line 259 of machtimer_alarm.c to:

```c
    } else if (self->handler != mp_const_none) {
```

This is done in master now. Will be available with the next release.

rkeiii, 2017-12-27:
@danicampora Thank you for the fix.

closed 2017-12-27 by rkeiii