pycom/pycom-micropython-sigfox #117: Obsolote stuff in the repository

robert-hh, 2018-01-01:
This repository contains a lot of stuff, which is not needed, just to mention a few of the TLD's:
bare-arm
cc3200
docs
esp8266
examples
logo
minimal
pic16bit
qemu-arm
stmhal
teensy
unix
windows
zephyr
They could be removed.

Xykon, 2018-01-02:
Some of these can definitely be removed but I believe we are using some code from the stmhal directory. I'll talk to @danicampora to see what he thinks.

robert-hh, 2018-01-02:
I made a test-clean and build (for LoPy) yesterday. There are indeed a few files in stmhal and a the sx127x subdir in drivers which were needed. Some others, like in examples or drivers/display, may still be useful. But without verification they create more confusion than help.

Xykon, 2018-06-12:
Sorry it took a while to get back on this, I had a discussion with Daniel at the time and we decided to keep the entire micropython project as is. You should consider this as a snapshot of the official micropython repository with modifications made to support our boards, rather than a firmware specifically for our (or generic) esp32 boards.

robert-hh, 2018-06-12:
That may be ok. But the question is: snapshot of what? The micropython.org repository has developed a lot since then, and there is also the Loboris branch, which is widely used. So effectively it is a firmware version specifically for Pycom's boards.

closed 2018-06-12 by robert-hh