pycom/pycom-micropython-sigfox #119: os.uname() causes core dump using firmware 1.1.12b on fopy

taxinvoice, 2018-01-07:
os.uname() causes core dump using firmware 1.1.12b on fopy

robert-hh, 2018-01-07:
That is already fixed in the latest build and should be gone, if you update the device.

pkourany, 2018-01-08:
Not fixed.  Just upgraded FiPy to 1.12.0.b1 and got  big dump:
```
>>> os.uname()
(sysname='FiPy', nodename='FiPy', release='1.12.0.b1', version='v1.8.6-849-fcbbfe3 on 2018-01-03', machine='FiPy with ESP32', lorawan='1.0.0', Guru Meditation Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400ea0bf  PS      : 0x00060330  A0      : 0x800ea374  A1      : 0x3ffd7560
A2      : 0x656d64a1  A3      : 0x3f4098bd  A4      : 0x00000000  A5      : 0x3ffd75e0
A6      : 0x00000000  A7      : 0x00000002  A8      : 0xd536f8d4  A9      : 0x000004d3
A10     : 0x00000002  A11     : 0x3f4098bb  A12     : 0x00000002  A13     : 0x3ffd75e0
A14     : 0x400f89c4  A15     : 0x00000001  SAR     : 0x0000001e  EXCCAUSE: 0x0000001c
EXCVADDR: 0xd536f8e4  LBEG    : 0x40098f9d  LEND    : 0x40098fad  LCOUNT  : 0xffffffff

Backtrace: 0x400ea0bf:0x3ffd7560 0x400ea371:0x3ffd7580 0x400eadf9:0x3ffd75a0 0x400eaf71:0x3ffd7600 0x400f255b:0x3ffd7650 0x400f2589:0x3ffd7670 0x400f1817:0x3ffd7690 0x400f97c0:0x3ffd76b0 0x400f3945:0x3ffd76e0 0x400f038d:0x3ffd7700 0x400fb5ef:0x3ffd7720 0x400f3a54:0x3ffd77c0 0x400f038d:0x3ffd7840 0x400f03ba:0x3ffd7860 0x400dda87:0x3ffd7880 0x400ddd0c:0x3ffd7920 0x400dcc81:0x3ffd7960

================= CORE DUMP START =================
xC0AAA0AAABsAQAA
LEb8P6B0/T8Ee/0/
kHL9P6B6/T9vEQEAGEP8PxhD/D8sRvw/EEP8PxQAAAD8YPs//GD7PyxG/D8AAAAA
BQAAAAhb/T9NaWNyb1B5AAAAAAAAAAAAAAAAAAR7/T8AAAAAIA0GAAUAAAABAAAA
AAAAAARP/D8AAAAAAAAAAAAAAADcT/0/RFD9P6xQ/T8AAAAAAAAAAAEAAAAAAAAA
fVJAPwAAAACwkglAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAgAAAA==
AQAAAL+gDkAwAwYAdKMOgGB1/T+hZG1lvZhAPwAAAADgdf0/AAAAAAIAAADU+DbV
0wQAAAIAAAC7mEA/AgAAAOB1/T/EiQ9AAQAAAB4AAAAcAAAA5Pg21Z2PCUCtjwlA
/////wIAAADgdf0/fCwIQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAPBSEIBwdf0/u5hAP72YQD/8rQ6AgHX9P3RpbWWwdf0/
4IsbgJB1/T8AAAAAu5hAP3SvDoCgdf0/LMhAPzB2/T9xAAAAAAAAACAAAAAAAAAA
XiUPgAB2/T8syEA/YJpAP9WJD4Dgdf0/0FFBP7uYQD/cUhBA5HCBPycAAAAAAAAA
MHb9PxB2/T8IAAAAAAAAAP////8AAAAAIAAAAAAAAAAAAAAAEHb9P2GaQD8MAAAA
jCUPgFB2/T8syEA/1C1APzB2/T8Qdv0/CAAAALuYQD8CAAAAAP8AAHRpbWUAAAAA
AgAAAKB3/T8syEA/0C1AP3RpbWUAAAAAAgAAAKB3/T8aGA+AcHb9PyzIQD8gLUA/
IC1APwcAAABALUA/IHj9P8OXD4CQdv0/LMhAPyAtQD8BAAAAAAAAACoAAABAeP0/
SDkPgLB2/T8gLUA/1K1APwEAAAABAAAAAAAAAA4OAACQAw+A4Hb9PyjBQD8BAAAA
AgAAAAAAAAAh9PgA7iwIQCzIQD8BAAAAAAAAANIKAADytQ+AAHf9PyjBQD8BAAAA
AAAAAOB3/T8AAAAAUAAAAFc6D4Agd/0/BAAAAOpwgT8AAAAA4Hf9P0h3/T+wd/0/
kAMPgMB3/T9IT/w/SE/8P4B4/T9Qd/0/VzoPgCB3/T+AeP0/AwAAAOh3/T8AAAAA
5Hf9PwAAAAAgeP0/gHGBPwEAAAB1AQAAkAMPgMB3/T9IT/w/SE/8PyKpD4CQcYE/
AwAAAAgAAADcd/0/AAAAAAAAAADAd/0/6Hf9P+R3/T8CAAAAAAAAACB4/T+QcYE/
3Hf9PwAAAAAw////63CBP70DD4BAeP0/MHCBPwAAAADWcIE/6XCBPwAAAADYd/0/
3Hf9P0hP/D8DAAAAKMFAPyAtQD8AAAAAzHj9PzBwgT8P3Q2AgHj9PwAAAAAQAAAA
0nCBPzB4/T9QeP0/AAAAAAAAAAAAAAAAAAAAABAAAADAd/0/AAAAAAMAAAAAeP0/
itoNgGB4/T8wcIE/AAAAAAAAAAAAAAAAAAAAAAAAAAAP3Q2AgHj9PzBwgT8QAAAA
AAAAAAAAAAAAAAAAAgAAAITMDYAgef0/AAAAAAAAAAAAAAAAsHj9Pw/dDYCAeP0/
AAAAAAAAAAAGAAAAAAAAACIAAAD//wCADQAAACRL/D8NAAAA/////4TMDYAgef0/
AAAAAAAAAAA32g2AkHGBP1BxgT8Aef0/IHn9PyofQD8AAAAAJEv8PwoAAAAAAAD/
BgAAAHMRAQAwcIE/AAAAAAYAAAAAAAAAIgAAAP//AIAAAAAAYHn9PwAO/D8BAAAA
IAAAAAoAAAAQcIE//v8AgAAAAABgef0/ZhMAAARP/D/weP0/AAAAAAAAAAB8S/w/
AAAAAKB6/T8AAAAAAAAAAAEAAAClpaWlpaWlpaWlpaXBAgAAmEsDeSBCT09ULlBZ
AKWlpaWlYm9vdC5weQClpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWl
paWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWlpaWl
zCwIQHDJDUAwAAUAAAAAAKB6/T8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAQAAAAAAAAD/////
AAAAAP/////////////////////FQhOBAAAAAAAAAAAAAAAAAAAZAGB5/T8AAAAA
AAAwAAAAAAAAAAAAAAAAAPMdQD/gR/w/AQAAANgdQD8AAAAAAAAAAAAAAAAAAAAA
AQAAAAEAAACsev0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAA==
bDD9PwAv/T9kMP0/
AC/9PwAw/T/3dRcVtEL8P7RC/D9sMP0/rEL8PxkAAABbnWSFoEtL2Www/T8AAAAA
AAAAAGgs/T9JRExFAPeDe23dFnTKN0sAAAAAAGQw/T8AAAAAIQAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADcT/0/RFD9P6xQ/T8AAAAAAAAAAAEAAAAAAAAA
fVJAPwAAAACwkglAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAKdp2Q==
zCwIQKhUFEAwBQYAErIIgMAv/T8gAAAAAQAAAAAAAACkj/w/////fwAAAADAj/w/
[... 203 lines cut ...]
AAAAAMUuCEAAAAAA2L8IQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAZAAAApaWlpaWlpaVA0A2AUMT9P9BN/D+wxP0/
AAAAACIAAAClpaWlpaWlpf////9QxP0/rFD8P7DE/T8AAAAAAQAAAMRN/D9cJvw/
AAAAAODE/T/ETfw/AAAAAADF/T8AAAAAAAAAAAAAAAAAAAAAxE38PwAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAABIT/w/
SE/8PwAAAACwxP0/ABQAAAAAAAAAAAAAAAAAAAAAAABIT/w/+Kv9P////38AAAAA
AAAAAADF/T8AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAMxf0/AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAA==
================= CORE DUMP END =================
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9028,len:8
load:0x3fff9030,len:1060
load:0x4009fa00,len:0
ho 12 tail 0 room 4
load:0x4009fa00,len:15160
entry 0x400a0644
```

gbeardall, 2018-01-11:
Same on Lopy4 with v1.13.0.b1:
```
>>> os.uname()
(sysname='LoPy4', nodename='LoPy4', release='1.13.0.b1', version='v1.8.6-849-3d5
56fb on 2018-01-10', machine='LoPy4 with ESP32', lorawan='1.0.0', Guru Meditatio
n Error of type LoadProhibited occurred on core  0. Exception was unhandled.
Register dump:
PC      : 0x400e9fc7  PS      : 0x00060030  A0      : 0x800ea27c  A1      : 0x3f
fd7200
A2      : 0x656d64a3  A3      : 0x3f4098b9  A4      : 0x00000000  A5      : 0x3f
fd7280
A6      : 0x00000000  A7      : 0x00000002  A8      : 0xd536f8dc  A9      : 0x00
0004d1
A10     : 0x00000002  A11     : 0x3f4098b7  A12     : 0x00000002  A13     : 0x3f
fd7280
A14     : 0x400f88d0  A15     : 0x00000001  SAR     : 0x0000001e  EXCCAUSE: 0x00
00001c
EXCVADDR: 0xd536f8ec  LBEG    : 0x40099105  LEND    : 0x40099115  LCOUNT  : 0xff
ffffff

Backtrace: 0x400e9fc7:0x3ffd7200 0x400ea279:0x3ffd7220 0x400ead01:0x3ffd7240 0x4
00eae79:0x3ffd72a0 0x400f2467:0x3ffd72f0 0x400f2495:0x3ffd7310 0x400f1723:0x3ffd
7330 0x400f96cc:0x3ffd7350 0x400f3851:0x3ffd7380 0x400f0295:0x3ffd73a0 0x400fb4f
b:0x3ffd73c0 0x400f3960:0x3ffd7460 0x400f0295:0x3ffd74e0 0x400f02c2:0x3ffd7500 0
x400dd9f7:0x3ffd7520 0x400ddc7c:0x3ffd75c0 0x400dcbf1:0x3ffd7600

..........
```

robert-hh, 2018-01-12:
I see an inconsistency here in moduos.c, line 232 (the line makred with !!!), but don't have an LOPY4 for testing. LOPY4 supports sigfox too.
```
STATIC const qstr os_uname_info_fields[] = {
    MP_QSTR_sysname, MP_QSTR_nodename,
    MP_QSTR_release, MP_QSTR_version
    ,MP_QSTR_machine
#if defined(LOPY) || defined(LOPY4) || defined(FIPY)
    ,MP_QSTR_lorawan
#endif
#if defined(SIPY) || defined(FIPY)   // !!!! Shouldn't be LOPY4 added here too?
    ,MP_QSTR_sigfox
#endif
}
```

robert-hh, 2018-01-12:
Confirmed: appening 
`|| defined(LOPY4)`
to line 232 of moduos.c fixes the core dump issue. Tested with a FIPY board.

robert-hh, 2018-01-17:
@pkourany That's fixed in 1.14.0.b1, and so the issue can be closed.

closed 2018-06-12 by Xykon