pycom/pycom-micropython-sigfox #120: Pytrack GPS timeout

thomasliebig, 2018-01-08:
I'm running MicroPython v1.8.6-849-fcbbfe3 on 2018-01-03; FiPy with ESP32 with a 
Pytrack (pytrack_0.0.8.dfu).

GPS just returns timeout with this script

```python
from L76GNSS import L76GNSS
from pytrack import Pytrack
import gc

py = Pytrack()
l76 = L76GNSS(py, timeout=60)
print ("Starting GPS")
while(True):
     coord = l76.coordinates(debug=True)
     print("{}".format(coord))
     gc.collect()
```

In flash/lib directory I have the attached py files [pytrack-lib.zip](https://github.com/pycom/pycom-micropython-sigfox/files/1611917/pytrack-lib.zip)
(These are not the latest ones, as the current pytrack.py using pycproc.py gave some error)
* pytrack.py
* LIS2HH12.py
* L76GNSS.py

danieleftodi, 2018-02-16:
try:
l76 = L76GNSS(py, timeout=10)

danieleftodi, 2018-02-16:
and add a "sleep" after gc.collect() ... like

gc.collect()
time.sleep(2)