pycom/pycom-micropython-sigfox #126: PyCom Firmware update not working under Linux 16.04

cparrish60, 2018-01-18:
Please include the following information when submitting a bug report:

LoPy
Trying to update to Firmware version 1.1.6.b01
I think the documentation is out of date.  Do I really need to run Python2?  I get an error saying Serial.py library not installed though I just installed it.  The documentation is awful.  The web link for Linux downloads the *.tar.gz.  What then?  I tried the manual method for installing previous versions (for current version) and get the Serial.py library error. 
Ideas?

Thank you!

sdaniel55, 2018-09-03:
Hi @cparrish60 👋 

Are you still having issues?

cparrish60, 2018-09-03:
I moved onto Arduino with a Multitech LoRa module.