pycom/pycom-micropython-sigfox #127: Sigfox socket.send appears to have stopped working on lopy4 with 1.14.0.b1

gbeardall, 2018-01-22:
(sysname='LoPy4', nodename='LoPy4', release='1.14.0.b1', version='v1.8.6-849-86d
a809 on 2018-01-17', machine='LoPy4 with ESP32', lorawan='1.0.0', sigfox='1.0.1')

With sigfox quick usage sample: https://docs.pycom.io/chapter/firmwareapi/pycom/network/sigfox.html

The sample was working with release 1.13.0.b1. After upgrading to 1.14.0.b1 i have not been able to send any messages to sigfox. If i revert to 1.13.0.b1 messages get through again.

I have done a sigfox 'disengage sequence number', to no effect. I have also looked at the sigfox device event and message logs.
Not sure how to diagnose this futher, Has anyone seen a similar issue, any suggestions?

thanks, gavin

danicampora, 2018-01-22:
@gbeardall thanks for reporting this. We will have a look a sort it out ASAP.

gbeardall, 2018-01-29:
Fixed in 1.15.0.b1

closed 2018-01-29 by gbeardall