pycom/pycom-micropython-sigfox #130: LoPy won't upgrade or connect via serial after pysense upgrade to 0.0.8.dfu

aidex-hackability, 2018-01-26:
**LoPy SN 1711004932**
* The firmware version MicroPython v1.8.6-237-**_g9d21f17_** on 2016-12-15; LoPy with ESP32

Yesterday I struggled to update pysense SN 1714001136.
No success but I was able to connect via Com11 to the LoPy via tera-term and get above info.
Finally today (20+attempts)  it succeeded with response:
F:\Development\LoPy>dfu-util-static.exe -D pysense_0.0.8.dfu
dfu-util 0.8
Match vendor ID from file: 04d8
Match product ID from file: f011
Opening DFU capable USB device...
ID 04d8:f011
....
DFU mode device DFU version 0100
Device returned transfer size 64
Copying data from PC to DFU device
Download        [=========================] 100%        16384 bytes
Download done.
state(2) = dfuIDLE, status(0) = No error condition is present
Done!

I then tried updating LoPy with pycom_firmware_update_1.1.6.b0.exe and it won't connect.
Tried tera-term and com11 no-longer available.
Neither FileZilla nor Atom will connect to ftp port
pysense is detected by Windows10 device manager as libusbK  device  USB\VID_04D8 & **PID_F012**
driver version 3.0.7.0

**_How do I correct situation to move forward?_**

catalinio, 2018-01-28:
Hi aidex-hackability,
It seems to be a Windows 10 drivers problems. The fact that you see PID F012, means the pysense firmware is running in application mode, so the update was fine.
If you used Zadig tool (http://zadig.akeo.ie/), you might have installed the libusbk for the Pysense in normal mode (USB PID F012), you should try to install (in Zadig) "USB Serial CDC" driver.
As a note, libusbk was required for the USB updater of Pysense (that's USB PID F011). And installing it, it should not affect the normal assignment of F012 as USB-to-Serial converter.
All the best,

aidex-hackability, 2018-01-28:
**# Thank you that worked!**
I was able to upgrade and get to the following:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff9028,len:8
load:0x3fff9030,len:1060
load:0x4009fa00,len:0
ho 12 tail 0 room 4
load:0x4009fa00,len:14084
entry 0x400a05d4
MicroPython v1.8.6-849-86da809 on 2018-01-17; LoPy with ESP32
Type "help()" for more information.
>>>

closed 2018-01-28 by aidex-hackability

aidex-hackability, 2018-01-28:
Okay - Next Question - why won't the LoPY connect via the Pymakr in Atom?
Connecting on 192.168.4.1...
Connection error: Error: Login timed out

catalinio, 2018-01-29:
Hi,
In Atom, you need to write a small script to instruct Lopy to which wifi to connect, here is an example: https://docs.pycom.io/chapter/tutorials/all/wlan.html
How to upload scripts in Atom: https://docs.pycom.io/chapter/gettingstarted/installation/pymakr.html
After Lopy is connected to wifi, you can check the IP address using "print(wlan.ifconfig())" and connect to ftp(using a client like FileZilla): https://docs.pycom.io/chapter/toolsandfeatures/FTP.html
As a suggestion, we have a quite active forum, maybe there you can post your issues, and more people could benefit: https://forum.pycom.io. Nevertheless, here will be also answered.