pycom/pycom-micropython-sigfox #141: Lora. OTAA join request at TTN fails with V1.16.0.1

robert-hh, 2018-02-23:
Node: FiPy with V1.16.0.b1
Gateway: LoPy with V1.16.b1
Software: scripts from the exmples lib: pycom-libraries/examples/lorawan-nano-gateway/
Both the gateway and the node are proeperly registered at TTN. Keys and identifiers are set.
OTAA requets are forwarded by the gateway, the TTN server creates a response within 5 sec, which is forwarded to the node. The node does not confirm the reception.
With V1.15.0.b1 at the node OTAA joib works, but about 3 out of 4 messages from the node to the gateway are not forwarded by the gateway.
Using ABP join at the node with V1.16.0.b1 works, and all uplink messages are delivered, but there is some trouble with downlink messages, which arriver at the gateway but get lost at the node.

P.S.: Distance node/gateway < 5 m. Simple device-device communication tests work flawless.

robert-hh, 2018-02-24:
The RXWindow1 Timer is pretty unstable. Setting the RxError value to 100ms improves the situation by extending the time, at which a packet is expected, to 200ms.
Where: https://github.com/pycom/pycom-micropython-sigfox/blob/master/lib/lora/mac/LoRaMac.c#L2377

sidpat, 2018-02-25:
Same issue here. 
Node: LoPy-`(sysname='LoPy', nodename='LoPy', release='1.16.0.b1', version='v1.8.6-849-055728a on 2018-02-13', machine='LoPy with ESP32', lorawan='1.0.0')`
Gateway: Lopy-`(sysname='LoPy', nodename='LoPy', release='1.16.0.b1', version='v1.8.6-849-055728a on 2018-02-13', machine='LoPy with ESP32', lorawan='1.0.0')`

Just something I noticed after update was that os.uname() states lorawan version to be 1.0.0 and the 1.16.0b1 firmware notes states 

> esp32: Update LoRaWAN to 1.0.2.

robert-hh, 2018-02-25:
The lorawan version number is just some text in the pycom_version.h file. So updating it may simply have been forgotten. I'm not sure about the effects, but with the change above, extending the Receive window length from 20ms to 200ms, joins almost every time works on the first attempt, latest on the third attempt.
BUT: The otaa_node.py example then sends dummy messages. Only 1 of 4 are received by the gateway. But the messages are actually sent. I can pick up the RF-signal next to the nodes antenna.

robert-hh, 2018-02-25:
Just made a test. The receiver should be always enabled with "device_class=LoRa.CLASS_C". But that does not make any difference. So I guess there something wrong in the LORA protocol stack implementation.

robert-hh, 2018-02-25:
The problem is especially the combination of xxPy node and xxPy gateway. In my tests, the node most of the time opened the RxWindow1 up to 50 ms too early, and the gateway started sending a little not too late.  With the larger Rx window at the node, this is compensated. A more precise timing would be better.

robert-hh, 2018-02-27:
The problem seems to be at a Lopy used as gateway with the nonogateway example code. The reason in besign analyzed.´and hopefully fixed. At the moment it looks like tick_us() is not sufficiently precise.

robert-hh, 2018-02-28:
Solved: The culprit is ticks_us(), which used the rtc clock running on an non-crystal oscillator. Changing ticks_us() and the SF127x drivers to use the CPU tick counter solved the problem. There is still an offset of about a few ms, which I might investigate further.

sidpat, 2018-03-01:
Updated firmware to V1.17.0.b1 for both LoPy node and LoPy gateway, still the same issue.

@robert-hh Could you show what changes you made to the code to make it working? Thanks.

robert-hh, 2018-03-01:
This is the git diff log of the changes I made to v1.17.0.b1. It looks more than it is. Actually in total only three lines have to be changed, two lines (include statements) added and two deleted (extern defines, wrong style anyhow):
```
diff --git a/drivers/sx127x/sx1272/sx1272.c b/drivers/sx127x/sx1272/sx1272.c
index 34a9e26..baacd7b 100644
--- a/drivers/sx127x/sx1272/sx1272.c
+++ b/drivers/sx127x/sx1272/sx1272.c
@@ -33,6 +33,7 @@ Maintainer: Miguel Luis and Gregory Cristian
 #include "sx1272.h"
 #include "sx1272-board.h"
 #include "esp_attr.h"
+#include "esp32_mphal.h"
 
 /*
  * Local types definition
@@ -1054,8 +1055,6 @@ static IRAM_ATTR void SX1272OnDioIrq (void) {
     }
 }
 
-extern uint64_t system_get_rtc_time(void);
-
 IRAM_ATTR void SX1272OnDio0Irq( void )
 {
     volatile uint8_t irqFlags = 0;
@@ -1073,7 +1072,7 @@ IRAM_ATTR void SX1272OnDio0Irq( void )
                     int8_t snr = 0;
 
                     // Store the packet timestamp
-                    SX1272.Settings.LoRaPacketHandler.TimeStamp = system_get_rtc_time();
+                    SX1272.Settings.LoRaPacketHandler.TimeStamp = mp_hal_ticks_us();
 
                     // Clear Irq
                     SX1272Write( REG_LR_IRQFLAGS, RFLR_IRQFLAGS_RXDONE );
diff --git a/drivers/sx127x/sx1276/sx1276.c b/drivers/sx127x/sx1276/sx1276.c
index 4338482..97198ea 100755
--- a/drivers/sx127x/sx1276/sx1276.c
+++ b/drivers/sx127x/sx1276/sx1276.c
@@ -33,6 +33,7 @@ Maintainer: Miguel Luis, Gregory Cristian and Wael Guibene
 #include "sx1276.h"
 #include "sx1276-board.h"
 #include "esp_attr.h"
+#include "esp32_mphal.h"
 
 /*
  * Local types definition
@@ -1219,8 +1220,6 @@ static IRAM_ATTR void SX1276OnDioIrq (void) {
     }
 }
 
-extern uint64_t system_get_rtc_time(void);
-
 IRAM_ATTR void SX1276OnDio0Irq( void )
 {
     volatile uint8_t irqFlags = 0;
@@ -1238,7 +1237,7 @@ IRAM_ATTR void SX1276OnDio0Irq( void )
                     int8_t snr = 0;
 
                     // Store the packet timestamp
-                    SX1276.Settings.LoRaPacketHandler.TimeStamp = system_get_rtc_time();
+                    SX1276.Settings.LoRaPacketHandler.TimeStamp = mp_hal_ticks_us();
 
                     // Clear Irq
                     SX1276Write( REG_LR_IRQFLAGS, RFLR_IRQFLAGS_RXDONE );
diff --git a/esp32/mods/modutime.c b/esp32/mods/modutime.c
index 5d26ef4..4595233 100644
--- a/esp32/mods/modutime.c
+++ b/esp32/mods/modutime.c
@@ -153,14 +153,12 @@ STATIC mp_obj_t time_sleep_us(mp_obj_t arg) {
 STATIC MP_DEFINE_CONST_FUN_OBJ_1(time_sleep_us_obj, time_sleep_us);
 
 STATIC mp_obj_t time_ticks_ms(void) {
-    struct timeval tv;
-    gettimeofday(&tv, NULL);
-    return mp_obj_new_int_from_uint(((tv.tv_sec * 1000) + tv.tv_usec / 1000));
+    return mp_obj_new_int_from_uint(mp_hal_ticks_ms());
 }
 STATIC MP_DEFINE_CONST_FUN_OBJ_0(time_ticks_ms_obj, time_ticks_ms);
 
 STATIC mp_obj_t time_ticks_us(void) {
-    return mp_obj_new_int_from_uint(system_get_rtc_time());
+    return mp_obj_new_int_from_uint(mp_hal_ticks_us());
 }
 STATIC MP_DEFINE_CONST_FUN_OBJ_0(time_ticks_us_obj, time_ticks_us);
```

danicampora, 2018-03-02:
Thanks @robert-hh will fix as you suggest, with the addition of placing ``mp_hal_ticks_ms`` inside IRAM as well.

robert-hh, 2018-03-02:
`mp_hal_ticks_us` for IRAM, I suppose. `mp_hal_ticks_ms` may be useful too, but not that critical, if there has to be a tradeoff between IRAM space and response.

danicampora, 2018-03-02:
Yes ``mp_hal_ticks_us``

sidpat, 2018-03-05:
@robert-hh thanks, OTAA works fine now.

robert-hh, 2018-03-17:
Putting mp_hal_ticks_xx and the respective modules in modutime.c into IRAM is not very effective, at least not for LoRa. And for gettimeofday(): I had the impression that this one is located in internal ROM, which is anyhow directly accessibly.

danicampora, 2018-03-17:
> Putting mp_hal_ticks_xx and the respective modules in modutime.c into IRAM is not very effective, at least not for LoRa.

Why not?

robert-hh, 2018-03-17:
Just if this turns out to be a problem. It is practically only used in the busy waiting loop of the lora gateway, when it waits for the downlink message to be scheduled. And there the timing uncertainty of the other calls is much larger. For instance from calling socket send to the RF signal coming up takes at least 3.2ms. It is more important that no timer ticks get lost. But that is outside the ticks_xx call scope.
And yes, I made the test with the little program that crashed with 1.17.2. The result with and w/o IRAM is the same. For less tight loops it might be different.

hallard, 2018-03-18:
This issues does not seem to be solved on my side , just wanted to test something with BLE and lost all my dedicated trying to find why my Lopy was not able to connect anymore to TTN with OTAA. I needed to reverse to firmware V1.15 to get it working. Of course I used to test all of this:

- Firmware LoPy-1.17.2.b1 don't connect
- Firmware LoPy-1.17.1.b1 don't connect
- Firmware LoPy-1.16.0.b1 don't connect
- Firmware LoPy-1.15.0.b1 works

Above V1.15 of course I changed 
```python
lora = LoRa(mode=LoRa.LORAWAN)
```
to
```python
lora = LoRa(mode=LoRa.LORAWAN, region=LoRa.EU868)
```

Here is my connect code, may be I missed something ?

```python
pycom.heartbeat(False)

# Initialize LoRa in LORAWAN mode.
lora = LoRa(mode=LoRa.LORAWAN, region=LoRa.EU868)
#lora = LoRa(mode=LoRa.LORAWAN)

# create an OTAA authentication parameters
dev_eui = binascii.unhexlify(config.DEV_EUI)
app_eui = binascii.unhexlify(config.APP_EUI)
app_key = binascii.unhexlify(config.APP_KEY)
timetojoin = 0
pycom.rgbled(0xffa500)  # LoRa heartbeat LED on

# join a network using OTAA (Over the Air Activation)
lora.join(activation=LoRa.OTAA, auth=(dev_eui, app_eui, app_key), timeout=0)

# wait until the module has joined the network
while not lora.has_joined():
    time.sleep(1)
    timetojoin += 1
    print('Not yet joined...')

pycom.rgbled(0x00ff00)
print('Joined in', timetojoin, 's')
time.sleep(1)

pycom.rgbled(0xff0000)
```

robert-hh, 2018-03-23:
The basic timing error is fixed, there is still a variable delay depending on the length of the downlink message. Maybe better to open a new isseu for that.

closed 2018-03-23 by robert-hh

investigatronica, 2019-11-01:
>     * Firmware LoPy-1.15.0.b1 works
I tested up until version 1.20.1 and only 1.15.0.b1 works