pycom/pycom-micropython-sigfox #143: Lora: Inconsistency in sendung uplink messages

robert-hh, 2018-02-25:
There seems to be a dependency on whether uplink messages are seen be the gateway depending on how the node join was performed. A simple test:
a) ABP join. 33 of 33 messages are seen by the gateway and forwarded to TTN
b) OTAA join. 13 of 33 messages are seen by the gateway and forwarded to TTN

Node: FiPy, 1.16.0.b1, script otaa_node.py from the examples
Gateway: LoPy1, 1.16.0.b1, script nanogateway.py from the examples

Funny enough, the OTAA join request message from the node is always seen & forwarded.
The messages are definitely sent by the node. Distance node-gateway 5m, good antennas, rssi ~-40, snr 6-7

danicampora, 2018-02-25:
@robert-hh thanks for reporting this. The difference with OTAA actually makes sense if the network server (TTN) is adding channels and the node is accepting it, making the nano-gateway (which only operates in 1 channel) to not receive the messages sent on the new channels. 

The fix is to reject "New Channel Requests" made on the channels that have been manually removed. This was the case on our previous LoRaWAN stack, but something didn't go right when moving to the new 1.0.2 stack.

robert-hh, 2018-02-25:
@danicampora Thank you for the explanation and the feedback in general.

closed 2018-02-27 by robert-hh