pycom/pycom-micropython-sigfox #150: BTLE start scan fails 'OSError: operation already in progress'

edhowork, 2018-03-15:
LoPy on pymakr PCB 
(sysname='LoPy', nodename='LoPy', release='1.17.0.b1', version='v1.8.6-849-d0dc708 on 2018-02-27', machine='LoPy with ESP32', lorawan='1.0.2')

We have a stable running application using BTLE stack on LoPy.
It scans for a BTLE device, connects and subscribes to characteristic notifications.

Whenever the BTLE connection is lost (ie I remove the power of the BTLE device) our application detects the connection lost and restarts the bt-scan after some time. The start_scan() call fails and reports 'OSError: operation already in progress'.

Calling isscanning() reports False.
None of the deinit, init, or calling the class constructor Bluetooth() resolves the problem. 
Only forcing a reset, machine.reset() will revive the system. 

We like to force the btle stack to a initial state. 
Please help.

keton, 2018-04-06:
+1, same here #148

semireg, 2018-04-10:
Is this something you can replicate with a minimum of code? If so, please post a code snippet to recreate. I've tested disconnect behavior with WiPY 3.0 and both spontaneous disconnect (what I would call "disconnect with error" - by removing the peripherals battery) and graceful disconnect (what I would call "disconnect without error" - my peripheral features a power-off button) work OK and I can connect multiple times without issue. I provided a simple script in the above issue referenced by @keton.

edhowork, 2018-04-10:
The problem arose after updating firmware of the LoPy. The application code was already finished and I can't strip down to share. 
We fixed the problem by downgrading to Firmware LoPy_868-1.9.2.b2

husigeza, 2018-08-14:
Hello,
Thanks for reporting, the fix will be part of the next development version!

closed 2018-08-14 by husigeza