pycom/pycom-micropython-sigfox #151: LoPy: Crash while running  LoRa Nanogateway

robert-hh, 2018-03-18:
configuration:
LoPy1, with V1.17.0.b1, modified to call mp_hal_us() in ticks_us() and the sc127x drivers.
I run the gateway since about two weeks. It crashes after a while, sometimes after a few hours, sometimes after a few days. Adding a WDT did not help, since it runs into a hard error. The only thing that my help is an external watchdog.
```
[  3784.203] Pull ack
[  3789.216] Push ack
[  3809.204] Pull ack
[  3834.228] Pull ack
[  3849.205] Push ack
[  3859.206] Pull ack
abort() was called at PC 0x40083743 on core 1

Backtrace: 0x4008e277:0x3ffc1280 0x4008e2a8:0x3ffc12a0 0x40083743:0x3ffc12c0 0x4008384d:0x3ffc1300 0x40083944:0x3ffc1320 0x4014f4c5:0x3ffc1340 0x40083e36:0x3ffc1360 0x40084a5a:0x3ffc1390 0x40084c69:0x3ffc13c0 0x40083f2f:0x3ffc13f0 0x40083589:0x3ffc1410 0x4000bfed:0x00000000

================= CORE DUMP START =================
pC8AAA4AAABsAQAA
0Kn9P1Co/T/Iqf0/
UKj9P2Cp/T+bUsezAFD8P0yv/T/Qqf0/+E/8PxkAAAAz8i32TR4P7dCp/T8AAAAA
AAAAAMyl/T9JRExFADnr7KJcoBUPj2cAAAAAAMip/T8AAAAAIQAGAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAADYJv0/QCf9P6gn/T8AAAAAAAAAAAEAAAAAAAAA
3lhAPwAAAAAYrwlAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAANP+Mg==
ODMIQHDhFEAwDwYA3L4IgBCp/T8gAAAAAQAAAAAAAACMpvw/7FH8P5RP/D+opvw/
AQAAAAAAAAD///9/N6gIgPC5/T8DAAAAIwAGAAAAAAAQqf0/IAAAAAAAAAAAAAAA
AAAAADE1CEDwuf0/RM0IQGit+j8AAAAAAAAAAAAAAAD//z+zAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAP//P7MAAAAAAAAAAAAAAAAAAAAAMKn9PwAAAAAAAAAA
AAAAAAQZ/D/sUfw/lE/8PwAAAABgqf0/AAAAAAAAAAAEGfw/AAAAAAAAAAAAAAAA
AQAAAAAAAAAEAAAAAQAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAGyp/T8AAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA
AAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAA==
```

nshishir, 2018-03-18:
Hello,
I am having similar problem. Beginner here. 
I am trying the Nano gateway sample with fresh Pysense+Lopy4. Updated firmware, registered with TTN. 
Node -
>>> os.uname()
(sysname='LoPy4', nodename='LoPy4', release='1.17.2.b1', version='v1.8.6-849-1bf377b on 2018-03-16', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

Gateway -
>>> os.uname()
(sysname='LoPy4', nodename='LoPy4', release='1.17.2.b1', version='v1.8.6-849-1bf377b on 2018-03-16', machine='LoPy4 with ESP32', lorawan='1.0.2', sigfox='1.0.1')

On starting the gateway, TTN shows that the gateway is connected. I see Pull Ack/Push Ack on console
On starting the node, the console says "Not joined yet" continuously.
The TTN console for Gateway shows multiple Join Request/Join Response
No activity on TTN console for the device though it shows it as available
Soon the gateway console starts having core dumps

rst:0xc (SW_CPU_RESET),boot:0x16 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff8028,len:8
load:0x3fff8030,len:1984
load:0x4009fa00,len:0
load:0x4009fa00,len:18636
entry 0x400a05b4
Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited)
. Exception was unhandled.
Register dump:

robert-hh, 2018-03-18:
I have a slightly modified nanogateway.py here, that runs on my LoPy so far, now with 1.17.2.b1
https://github.com/robert-hh/pycom-libraries/blob/master/examples/lorawan-nano-gateway/nanogateway.py

nshishir, 2018-03-18:
Hello Robert,
Thanks for the quick response. I am using the code available at
https://github.com/pycom/pycom-libraries/blob/master/examples/lorawan-nano-gateway/nanogateway.py

I couldnt find any difference in the code in the above link. Am I missing anything?

Regards

robert-hh, 2018-03-19:
Sorry, wrong branch. This is the right one:
https://github.com/robert-hh/pycom-libraries/blob/nanogateway/examples/lorawan-nano-gateway/nanogateway.py

robert-hh, 2018-03-23:
Seems to be fixed with 1.17.3.b1

closed 2018-03-23 by robert-hh

nshishir, 2018-03-23:
Are you using your branch code for nanogateway or the master branch?
OTAA or ABP??

robert-hh, 2018-03-23:
No, I'm using my own version of the nanogateway lib, both OTAA and ABP. But this issue was about the crash (call for the GURU), not about the timing problems and not about lost uplink messages.