pycom/pycom-micropython-sigfox #153: AttributeError: 'poll' object has no attribute 'ipoll'

donikuy, 2018-03-23:
I am using a WIPY 3.0 on a WIPY 3.0. 
MicroPython v1.8.6-849-83e2f7f on 2018-03-19; WiPy with ESP32

When I try to use uasyncio module i get the following error:

Traceback (most recent call last):
File "main.py", line 80, in <module>
File "asyncio_priority.py", line 128, in run_forever
File "uasyncio/__init__.py", line 69, in wait
AttributeError: 'poll' object has no attribute 'ipoll'
MicroPython v1.8.6-849-83e2f7f on 2018-03-19; WiPy with ESP32

Would anyone be able to explain this error please?

Thank you!

Sympatron, 2018-03-23:
That's because poll.ipoll() is not implemented in the Pycom fork of MicroPython.

Your easiest option is probably to change the uasyncio source to use poll.poll() instead. It is basically the same but less memory efficient. But since the WiPy3 has 4MB RAM it shouldn't be a problem for you.

donikuy, 2018-03-24:
Thank you. I have changed it. I think I was also getting confused because my software reaches loop.run_forever(), but doesn't seem to run anything, does not even go to the first task.

donikuy, 2018-03-24:
Seems to get stuck in that loop:

Traceback (most recent call last):
  File "main.py", line 59, in <module>
  File "asyncio_priority.py", line 128, in run_forever
  File "uasyncio/__init__.py", line 73, in wait

husigeza, 2018-08-07:
Hello,
uasyncio support among with the ipoll() added in development version 1.19.0.b4 thus closing this issue.

closed 2018-08-07 by husigeza