pycom/pycom-micropython-sigfox #155: LoRa ischannel_free freezes LoPy

pgorczak, 2018-03-29:
When calling `ischannel_free`, the device freezes.

My system info:

* LoPy with ESP32
* release 1.17.3.b1
* version v1.8.6-849-83e2f7f on 2018-03-19

Script to reproduce

```Python
from network import LoRa
lora = LoRa(mode=LoRa.LORA, region=LoRa.EU868)
lora.ischannel_free(-100)
```

johanneskrauser2, 2018-07-25:
Hi pgorczak,
I seem to be having the same issue with a LoPy4. Did you possibly manage to find a solution?

pgorczak, 2018-08-03:
Hi @johanneskrauser2 unfortunately no but it's good to know I'm not alone :smile:

robert-hh, 2018-08-04:
I cannot confirm this. Using the code lines above, the call to is_channel_free() returns immediately on my LoPy4 (sx1276 chip) and FiPy (sx1272 chip). Firmware 1.18.0.r1. 
F.Y.I.: The internal wait time for the channel check is set to 2 ms.

johanneskrauser2, 2018-08-04:
I think it was a firmware issue. I don't recall the exact version, but when I purchased the devices I had done a firmware update to the latest development release. A few weeks later (when I encountered this issue) I did another firmware update to the latest stable release and it seems to have resolved the issue.

@robert-hh Thanks for the FYI. I was actually wondering about that. Do you know if its possible to examine the internal coding of the various class functions (such as is_channel_free()) so that I can better understand how they work at a lower level?

Thanks.

robert-hh, 2018-08-04:
If you loog in to the source tree, starting at pycom-micropython-sigfox (abbr. PMS), you'll find a subdirectory call esp32. That contains all esp32 specific code. in esp32/mods are the python interfaces for the various modules, like esp32/mods/modlora.c and .h for the Lora class. The LoRa support is quite dispersed in the sources, and it goes forth an back between the various places: 
PMS/esp32/lora        Pycom board specific adaption
PMS/drivers/sc27x   Semtech drivers for the LoRa chips
PMS/lib/lora/...          Semtech implementation of the Lora protocols