pycom/pycom-micropython-sigfox #156: Error Temperature Reading

GaneshVarahade, 2018-04-05:
* The Pycom board (FiPy, Pysense).
* Firmware version 1.17

* Exact steps to cause this issue
    1. I've run the FiPy on Pysense to get the environmental data readings to my website using MQTT protocol
    2. The temperature & humidity readings, I got were not the correct one's as compared to the real which I've got from another device.

I'm a developer to get these critical data to my website. Do help me out with the error correction in the environmental data readings that I get from the Pycom devices. Else there's any update in the coming firmware or the device.

Thank you!

robert-hh, 2018-04-05:
What do you get and what do you expect?

robert-hh, 2018-04-05:
To translate that into writen text: You expect 28.8 °C, and you get 32.16?
There was quite a bit of discussion in the pycom forum (forum.pycom.io) about this effect, like here. That's not a software issue, but a design flaw of the pysense PCB, picking up too much heat from the module and the board.

GaneshVarahade, 2018-04-05:
Please help me with the exact link where this issue has been discussed in the pycom forum.
Are the Pycom guys gonna fix this bug in the next PCB board?

robert-hh, 2018-04-05:
https://forum.pycom.io/topic/2001/pysense-accuracy
In this thread they tell, they will fix it.

closed 2018-09-03 by sdaniel55