pycom/pycom-micropython-sigfox #167: Expansion Board 3.0 not boot from battery

lancernz, 2018-05-29:
Pycom Expansion Board 3.0

(sysname='GPy', nodename='GPy', release='1.17.3.b1', version='v1.8.6-849-83e2f7f on 2018-03-19', machine='GPy with ESP32')

First I uploaded our latest code for the Gpy module - which works great when the Expansion Board is plugged into a USB 5V source.
Then with the battery plugged in I removed the 5V source (USB). Still works great.
Then I removed the battery, installed the board into a box, then plugged the battery back in.
No go.
I expected it would boot and run just as it does using the Expansion Board 2.0

Put 5V USB power in, boots again.

Then I tried the RGB test LED sketch.
Works great with 5V supply.
Remove the 5V supply and run from battery - still works great. 
Hit the reboot button, it won't boot.

What I also found was when plugging the battery in the USB LED glows (half brightness), which defeats being able to run the Expansion Board from battery in deep sleep for long periods (LED will drain the battery quite quickly).

I've not tripped upon any differences in the 2.0 to 3.0 that suggest the USB LED can be disabled or should be on when the battery is connected. Also, how can I boot the board with the battery as the primary power source?

Any ideas?

Thank you!

danicampora, 2018-05-29:
Hello,

Can you try connecting P8 to 3V3 and resetting the module?

Cheers,
Daniel

lancernz, 2018-05-30:
Yes I can confirm connecting P8 (G15) to 3V3 resolves this and it operates from battery.
Thank you.

closed 2018-05-30 by lancernz

AndrewStyan, 2019-02-09:
Just encountered this problem and has taken me a while to 'discover' this solution. Yes, it is on the expansion board info page, but I initially didn't believe it, thinking "well of course I want to use the battery ...  why do I have to commit a pin to this ? why is this necessary?".

It seems like a design flaw... will it actually get fixed at some point?

scienceintheshed, 2019-04-03:
I've tried this with a SiPy attached to an Expansion Board V3.1, both with the latest firmware and it does not work.