pycom/pycom-micropython-sigfox #169: Make Target-boot error

AKethamreddy, 2018-06-07:
Hi ,
We have tested the following configuration in both Ubuntu and MAC OS.
* FiPy
* Trying to update Firmware over USB/UART.
Using pycom-esp-idf as IDF path. 

I got the following error while trying to make BOARD-FIPY Target-boot. 

I got following error:
```
CC bootloader/bootmgr.c
CC bootloader/mperror.c
CC bootloader/gpio.c
CC bootloader/flash_qio_mode.c
In file included from /home/user/esp/pycom-esp-idf/components/esp32/include/rom/ets_sys.h:21:0,
                 from /home/user/esp/pycom-esp-idf/components/log/include/esp_log.h:21,
                 from bootloader/flash_qio_mode.c:17:
bootloader/flash_qio_mode.c: In function 'enable_qio_mode':
bootloader/flash_qio_mode.c:179:66: error: 'EFUSE_RD_CHIP_VER_RESERVE_S' undeclared (first use in this function)
         uint32_t chip_ver = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_RESERVE);
                                                                  ^
/home/user/esp/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:189:32: note: in definition of macro 'REG_GET_FIELD'
             ((REG_READ(_r) >> (_f##_S)) & (_f##_V));                                                                   \
                                ^
bootloader/flash_qio_mode.c:179:66: note: each undeclared identifier is reported only once for each function it appears in
         uint32_t chip_ver = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_RESERVE);
                                                                  ^
/home/user/esp/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:189:32: note: in definition of macro 'REG_GET_FIELD'
             ((REG_READ(_r) >> (_f##_S)) & (_f##_V));                                                                   \
                                ^
bootloader/flash_qio_mode.c:179:66: error: 'EFUSE_RD_CHIP_VER_RESERVE_V' undeclared (first use in this function)
         uint32_t chip_ver = REG_GET_FIELD(EFUSE_BLK0_RDATA3_REG, EFUSE_RD_CHIP_VER_RESERVE);
                                                                  ^
/home/user/esp/pycom-esp-idf/components/soc/esp32/include/soc/soc.h:189:44: note: in definition of macro 'REG_GET_FIELD'
             ((REG_READ(_r) >> (_f##_S)) & (_f##_V));                                                                   \
                                            ^
../py/mkrules.mk:47: recipe for target 'build/FIPY/release/bootloader/flash_qio_mode.o' failed
make: *** [build/FIPY/release/bootloader/flash_qio_mode.o] Error 1
```

Thank you!

Xykon, 2018-06-07:
Did you check out the correct IDF commit as indicated [here](https://forum.pycom.io/topic/3326/new-firmware-release-v1-18-0)?

AKROF, 2018-06-07:
Hi, 
That helped . Thank you.

closed 2018-06-07 by AKROF